lianghongzhuo/PointNetGPD #2: Generate dataset problem about python generate-dataset-canny.py [prefix]

RJ13012416, 2019-02-27:
Hi, Liang. I follow your instructions install the modified meshpy,
/home/a/图片/2.png
/home/a/图片/3.png
While the reselut about generate-dataset-canny.py is 
/home/a/图片/1.png

RJ13012416, 2019-02-27:
Hi, Liang.
Some package like matplotlib etc is now support Python3 and I see the code you publish in here is python2 , how do I slove the version conflicts?

lianghongzhuo, 2019-02-27:
@RJ13012416 
Hi,
1. I can not see the pictures, could you upload again?
2. In the code, only ROS related codes are using python2,  for example [this code](https://github.com/lianghongzhuo/PointNetGPD/blob/master/dex-net/apps/kinect2grasp_python2.py)
For the code to generate dataset and training, it's ok to install the code in python3. And we don't use matplotlib when running ROS codes. 
In short, the rest codes are compatible with both python2 and python3

RJ13012416, 2019-03-01:
@lianghongzhuo 
Thanks for your reply.

I had running in my computer, while the result of the program is always data about the first model 003_cracker_box 

[image]

[image]

[image]

Could you help me why this happened ?  and I haven't changed the program .

RJ13012416, 2019-03-01:
when running the code :

def worker(i, sample_nums, grasp_amount, good_grasp):
    object_name = file_list_all[i][len(home_dir) + 35:]
    print('a worker of task {} start'.format(object_name))

And  the output is :
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start
a worker of task 003_cracker_box start

RJ13012416, 2019-03-09:
I had slove it, Thx.

jeasinema, 2019-03-26:
Close due to no potential update.

closed 2019-03-26 by jeasinema

lesleyhaha, 2019-04-15:
> when running the code :
>
> def worker(i, sample_nums, grasp_amount, good_grasp):
> object_name = file_list_all[i][len(home_dir) + 35:]
> print('a worker of task {} start'.format(object_name))
>
> And the [...]

How did you solve this problem? I want to know the time you deal with your dataset. Thanks so much!

bdenoun, 2019-04-16:
@RJ13012416 could you please share how you fixed that? I am also interested. Thank you

GeorgeDu, 2019-07-11:
> I had slove it, Thx.

I met the same problem, and could you please share how you fixed that? Thanks.

GeorgeDu, 2019-07-15:
> > I had slove it, Thx.
> 
> I met the same problem, and could you please share how you fixed that? Thanks.

The pool_size could be set according to your machine. If it was set as 1, only one model will be processed.

TheJasal, 2019-07-27:
@lesleyhaha you should check the output be careful, and I adviced you print the code output setp by step, and then you will find how to mmodified.
Hope that helps!

TheJasal, 2019-08-10:
@jeasinema @lianghongzhuo _Hi_，I had meet a confused. The topic  **''/table_top"** and **"/table_top_points"** appeared in kinect2grasp_python2.py, I don't konw how they defined.
Would you kind to push the code in this github.
Thanks!