lianghongzhuo/PointNetGPD #3: How much time to generate dataset?

iory, 2019-03-27:
Thanks for your great works!

I would like to train PointNetGPD.
And data generation took much time and it is not finished yet.
I want you to tell me that how much time to generate a dataset  for reference.

lianghongzhuo, 2019-03-27:
Estimated running time at our Intel(R) Xeon(R) CPU E5-2690 v4 @ 2.60GHz dual CPU with 56 Threads is 36 hours. Please also remove objects beyond the capacity of the gripper.

iory, 2019-03-28:
Thank you!

>  Please also remove objects beyond the capacity of the gripper.

OK. I will retry data generation.

closed 2019-03-28 by iory

lesleyhaha, 2019-04-15:
> Estimated running time at our Intel(R) Xeon(R) CPU E5-2690 v4 @ 2.60GHz dual CPU with 56 Threads is 36 hours. Please also remove objects beyond the capacity of the gripper.

Thanks! I also has a question that  how much time to generate a object in the dataset, like 002_master_chef_can.

zhangcogito, 2019-11-30:
> > Estimated running time at our Intel(R) Xeon(R) CPU E5-2690 v4 @ 2.60GHz dual CPU with 56 Threads is 36 hours. Please also remove objects beyond the capacity of the gripper.
> 
> Thanks! I also has a question that how much time to generate a object in the dataset, like 002_master_chef_can.

Have you resolved it? I run "python generate-dataset-canny.py [prefix]". It start generating the grasp of 002_master_chef_can. It has run for several hours and still not stops. Why it costs so much time to generate grasps for only 1 object?

v-wewei, 2020-01-13:
@zhangcogito hi，did you solve the problem for the time consuming about generating grasp for objects?

toannguyen1904, 2023-07-13:
hello, any updates on the generation time?