lianghongzhuo/PointNetGPD #4: No module named main_test

prakveer, 2019-04-01:
I tried to run kinect2grasps.py file and  can't import main_test. 
Other modules import correctly, any help is appreciated.
Thanks in advance.

lianghongzhuo, 2019-04-09:
Hi, 
There was a code path change for `PointNetGPD`, I updated the code. Please pull the newest code to fix this.

bdenoun, 2019-04-16:
Hi,

I just cloned the repository and made a ctrl+shift+f and no main_test file has been found

lianghongzhuo, 2019-04-16:
@prakveer  @bdenoun

prakveer, 2019-04-16:
Can you also upload the pretrained models ?
torch fails to load the default model: '~/code/grasp-pointnet/PointNetGPD/data/pointgpd_chann3_local.model'
Thanks!

lianghongzhuo, 2019-04-17:
@prakveer For the model, I suggest to train it yourself. :smile:

prakveer, 2019-04-17:
Cool. I'll give it a try :sweat_smile: