lianghongzhuo/PointNetGPD #7: bug when running python kinect2grasp_python2.py

TheJasal, 2019-08-05:
Traceback (most recent call last):
  File "kinect2grasp_python2.py", line 51, in <module>
    from main_test import test_network, model, args
  File "/main_test.py", line 50, in <module>
    model = torch.load(args.load_model, map_location='cpu')
  File "/usr/local/lib/python2.7/dist-packages/torch/serialization.py", line 358, in load
    return _load(f, map_location, pickle_module)
  File "/usr/local/lib/python2.7/dist-packages/torch/serialization.py", line 542, in _load
    result = unpickler.load()
AttributeError: 'module' object has no attribute '_rebuild_parameter'

how to fix this problem?thanks

ruanjian0821, 2019-08-05:
This may be helpful.
https://github.com/NVIDIA/tacotron2/issues/182

closed 2019-08-05 by TheJasal

TheJasal, 2019-08-05:
using torch 1.1.0 and referencing the answer by @ruanjian0821 solved the problem.
new issue:
Traceback (most recent call last):
  File "kinect2grasp_python2.py", line 537, in <module>
    if_good_grasp, grasp_score_tmp = test_network(model.eval(), points_modify)
  File "main_test.py", line 71, in test_network
    output, _ = model_(local_pc)  # N*C
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "pointnet.py", line 186, in forward
    x, trans = self.feat(x)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "pointnet.py", line 137, in forward
    trans = self.stn(x)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/robot/code/grasp-pointnet/PointNetGPD/model/pointnet.py", line 30, in forward
    x = F.relu(self.bn1(self.conv1(x)))
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/conv.py", line 196, in forward
    self.padding, self.dilation, self.groups)
RuntimeError: Calculated padded input size per channel: (0). Kernel size: (1). Kernel size can't be greater than actual input size.

ruanjian0821, 2019-08-06:
Checking the "local_pc" shape.
It's looks like there is a problem with the input.

TheJasal, 2019-08-06:
problem solved.
we use pointgpd_chann3_local.model as trained model
in main_test.py
change parser.add_argument('--model_type', type=str) to parser.add_argument('--model_type', type=str,default='100').

ruanjian0821, 2019-08-06:
good news！

TheJasal, 2019-08-09:
[image]

TheJasal, 2019-08-09:
The calculated marker array is really high as the picture shows.
I don't know why.
I am sure the frames 'world','table_top' and 'kinect2_ir_optical_frame' are set correctly. 
I checked the value of 'position' in real_good_grasp,and its z value is always 2 to 4,much higher than the ground.

ruanjian0821, 2019-08-09:
Find in dex-net/apps/kinect2grasp_python2.py, some code may be helpful.

# as the kinect2 is not well colibrated, here is a work around
points_[:, 0] = points_[:, 0] + 0.025  
points_[:, 2] = points_[:, 2] #+ 0.018

TheJasal, 2019-08-10:
The table_top_points and table_top topic don't know how to defined!

trungpham2606, 2020-10-07:
@ruanjian0821 
@TheJasal 
Can you guys send me the weight: pointgpd_chann3_local.model ?
The new weight that the author provide seems not working.

ACaseOfOrange, 2020-10-24:
> The calculated marker array is really high as the picture shows.
> I don't know why.
> I am sure the frames 'world','table_top' and 'kinect2_ir_optical_frame' are set correctly.
> I checked the value of 'position' in real_good_grasp,and its z value is always 2 to 4,much higher than the ground.

how did you set 'table_top' and 'kinect2_ir_optical_frame'? Thanks!

Twilight89, 2021-03-09:
@TheJasal  Hi~ Do you find the solution? I'm trouble in the same problem...