lianghongzhuo/PointNetGPD #11: Can't generate grasps for 002_master_chef_can.

zhangcogito, 2019-11-30:
Hi, Liang,
I follow the instruction and I just put just one model 002_master_chef_can in the ~/dataset/ycb_meshes_google/objects. Then I run

> $ python generate-dataset-canny.py 

 I got these warning:
```
$ python generate-dataset-canny.py bit
WARNING:root:Failed to import geometry msgs in rigid_transformations.py.
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
WARNING:root:Unable to import CNN modules! Likely due to missing tensorflow.
WARNING:root:TensorFlow can be installed following the instructions in https://www.tensorflow.org/get_started/os_setup
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Failed to import ROS in Kinect2_sensor.py. Kinect will not be able to be used in bridged mode
WARNING:root:Unable to import Primsense sensor modules! Likely due to missing OpenNI2.
WARNING:root:Unable to import pyrealsense2.
WARNING:root:Failed to import ROS in ensenso_sensor.py. ROS functionality not available
WARNING:root:Failed to import ROS in phoxi_sensor.py. PhoXiSensor functionality unavailable.
WARNING:root:Unable to import generic sensor modules!.
WARNING:root:Unable to import weight sensor modules!
WARNING:root:Failed to import gqcnn! Grasp2D functions will not be available.
WARNING:root:Failed to import OpenRAVE
WARNING:dexnet.grasping.grasp_sampler:Failed to import OpenRAVE
WARNING:dexnet.grasping.grasp_sampler:Failed to import rospy, you can't grasp now.
All job done.
a worker of task 002_master_chef_can start
```

Then, it runs for several hours withou ending. And the output repeated 
 ```
INFO:dexnet.grasping.grasp_sampler:47/100 grasps found after iteration 2.
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
INFO:dexnet.grasping.grasp_sampler:31/100 grasps found after iteration 1.
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
```
All of the warnings should be fixed? But they need ros and openrave. ROS and epenrave depend on python2, but you said the code were written in python3. How could I resolve these problems?

lianghongzhuo, 2019-12-04:
These warnings are for dex-net, you can ignore it. 
Openrave, pylibfreenect2, gqcnn are not needed for this project.

v-wewei, 2019-12-31:
@zhangcogito hi, did you finally solve the problem and get the grasp ? i have the same question with you.

zhangcogito, 2020-01-10:
> @zhangcogito hi, did you finally solve the problem and get the grasp ? i have the same question with you.

最后还是没弄出来，配置3D的环境本来就麻烦，再加上这个代码库还是python2 python3混着用，实在没办法，放弃了。

Hymwgk, 2020-08-01:
@zhangcogito   我现在也在跑这个，电脑是双E5-2699 v3共72线程，但是跑002_master_chef_can跑了一个小时，还在继续，不知道怎么解决，不知道是哪里出了问题

Hymwgk, 2020-08-01:
@zhangcogito 兄弟你搞出来了嘛？

lianghongzhuo, 2020-08-01:
默认设置是跑出7000个抓取再保存，你可以先把生成的抓取个数设置少一些

lianghongzhuo, 2020-08-03:
You can download the dataset from: https://tams.informatik.uni-hamburg.de/research/datasets/PointNetGPD_grasps_dataset.zip

closed 2020-08-03 by lianghongzhuo

lochinasc1017, 2020-08-18:
> 默认设置是跑出7000个抓取再保存，你可以先把生成的抓取个数设置少一些
您好，感谢您非常出彩的工作，有两个问题请教下
1. 哪里可以设置抓取个数 7000?，设置少一些是不是就可以生成快一些，因为我想通过这种方式看看数据集生成后的结果是什么样的？
2.PointNetGPD下这两个文件夹 ycb_grasp 和 ycb_rgbd 需要自己创建，连接时，我的dataset并没有这两个文件夹，想问下这两个文件夹是在哪一步生成的？跟您提供的可下载数据集有关系吗？可否详细描述下您提供的数据集放到哪个文件夹？

非常感谢！！

├── google2cloud.csv  (Transform from google_ycb model to ycb_rgbd model)
├── google2cloud.pkl  (Transform from google_ycb model to ycb_rgbd model)
├── ycb_grasp -> $HOME/dataset/ycb_grasp  (Links to the dataset directory)
├── ycb_meshes_google -> $HOME/dataset/ycb_meshes_google/objects  (Links to the dataset directory)
└── ycb_rgbd -> $HOME/dataset/ycb_rgbd  (Links to the dataset directory)

lianghongzhuo, 2020-08-18:
@lochinasc1017 你可以试着改一下 https://github.com/lianghongzhuo/PointNetGPD/blob/master/dex-net/apps/generate-dataset-canny.py#L36-L37
这里的20 和50的值

pierhui, 2021-05-07:
你好，请问你知道数据集生成后是什么结果吗？提供的数据集我们应该放在哪里呢？

pierhui, 2021-05-07:
@lochinasc1017 你好，你的这两个问题解决了吗，想向你请教一下。

lianghongzhuo, 2021-05-07:
@pierhui You can try the new code, and modify [this line](https://github.com/lianghongzhuo/PointNetGPD/blob/master/dex-net/apps/generate-dataset-canny.py#L37) to a smaller value.