lianghongzhuo/PointNetGPD #12: No module named msg.gpd when running kinect2grasp

PCIHD, 2019-12-15:
File "kinect2grasp_python2.py", line 15, in <module>
    from gpd.msg import GraspConfig
ImportError: No module named gpd.msg
Please help.

lianghongzhuo, 2019-12-15:
This file use the gpd grasp message to publish grasp messages. To solve this, you can do the following:
Method 1: Install gpd_ros and gpd package, https://github.com/atenpas/gpd_ros, https://github.com/atenpas/gpd
Method 2: Put the message file https://github.com/atenpas/gpd_ros/blob/master/msg/GraspConfig.msg
and https://github.com/atenpas/gpd_ros/blob/master/msg/GraspConfigList.msg to your own ROS package, and modify File "kinect2grasp_python2.py", line 15 and line 16 to:
`from YOUR_PACKAGE_NAME.msg import GraspConfig` and
`from YOUR_PACKAGE_NAME.msg import GraspConfigList`

PCIHD, 2019-12-16:
That worked , opted of option one , only catch is the catkin workspace must be sourced initially.

closed 2020-08-03 by lianghongzhuo