lianghongzhuo/PointNetGPD #14: Problem about python generate-dataset-canny.py

v-wewei, 2019-12-31:
Hi, @lianghongzhuo 
i have a question about the code for Generate dataset problem about python generate-dataset-canny.py . 
first , it seems the code hard code the pool_size to be 1, do we need to modify this setting to the actual cores for our cpu cores?
second, in line 36 "p_set = [multiprocessing.Process(target=worker, args=(i, 100, 20, good_grasp)) for _ in range(50)] " , i just got confuse where would this line to fill the list with the same args for the following 50 processes ? Seems the total 50 processes to do the same thing , right ? or is there any random variable for the 50 processes?
Thanks advance and looking forward your reply !
Much thanks for your great work for robot grasping .

lianghongzhuo, 2020-01-13:
The dataset generation script require a powerful server with multiple CPUs. If you really want some dataset, I can help you generate a dataset on our server when it is free. 
[image]

v-wewei, 2020-01-14:
@lianghongzhuo 
Yes, Thanks. i have notice that not all of the object in ycb_video are choosed in dataset generation, would you like to give a list of the object set for dataset generation ? or does it the same as the list in  
[image]

lianghongzhuo, 2020-01-14:
You are right, I only generate a dataset with objects on this list.
The reason is:
1. some of the objects have poor pointcloud data
2. some of the objects do not have pointcloud or depth data
3. some fo the objects are too big that exceed the gripper I use.

v-wewei, 2020-01-16:
@lianghongzhuo since i would like to follow your work pointnetgpd . But some of the config or parameters do not share a clear definetion for the gripper . 
would you like to give a detail explainment for these parameters as follow:

"finger_width":
"real_finger_width":
"hand_height":
"hand_height_two_finger_side":
"hand_outer_diameter":
"hand_depth":
"real_hand_depth":
"init_bite":

this parameter quite confuse me a lot in your configuration . if not time consuming , maybe you can provide a figure for the detail information.

here we would like to provide a figure for our two finger gripper . 
[image]

i can also send you the obj file if provided your email address.
Thanks very much for your reply !

lianghongzhuo, 2020-01-17:
These parameters have the same definition as of [gpg](https://github.com/atenpas/gpg) package.
The defination is [here](https://github.com/atenpas/gpg/blob/master/cfg/params.cfg):
```
# Robot Hand Geometry Parameters:
#   finger_width: the width of the finger
#   outer_diameter: the diameter of the robot hand (= maximum aperture plus 2 * finger width)
#   hand_depth: the finger length (measured from hand base to finger tip)
#   hand_height: the height of the hand
#   init_bite: the minimum distance between the fingertip and the side of the object that is oriented toward the hand
```
I think it's quite easy to get the idea.

Ctiger96, 2020-10-20:
@v-wewei Hi, do you know the meaning of "finger_radius"?

closed 2020-12-14 by v-wewei