lianghongzhuo/PointNetGPD #15: Are the weights provided?

clemense, 2020-01-27:
Hi,
I can't find the trained NN weights. Are they not provided?
Thanks!

lianghongzhuo, 2020-01-27:
Yes, weights file is located at https://github.com/lianghongzhuo/PointNetGPD/blob/master/data/pointgpd_3class.model

bdenoun, 2020-02-06:
Hi,

When trying to run the provided weights this is what I get:

Traceback (most recent call last):
  File "main_test.py", line 106, in <module>
    main()
  File "main_test.py", line 97, in main
    predict.append(test_network(model, local_pc).item())
  File "main_test.py", line 72, in test_network
    output, _ = model_(local_pc)  # N*C
  File "/home/user/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/user/.local/lib/python2.7/site-packages/torch/nn/parallel/data_parallel.py", line 150, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/home/user/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/user/.local/lib/python2.7/site-packages/torch/nn/parallel/data_parallel.py", line 143, in forward
    if t.device != self.src_device_obj:
  File "/home/user/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 576, in __getattr__
    type(self).__name__, name))
AttributeError: 'DataParallel' object has no attribute 'src_device_obj'

Any ideas how to fix this?

closed 2020-02-07 by lianghongzhuo