lianghongzhuo/PointNetGPD #17: Issue when runninf main_test.py

bdenoun, 2020-02-06:
Hi,

I have tried to run the provided neural network using both kinect2grasp_pyhton2.py and just main_test.py but in both scenario I have an error stating that: 

AttributeError: 'DataParallel' object has no attribute 'src_device_obj'

I have tried to comment out everything related to GPUs in the code, but doesn't change anything....

Any help would be appreciated.

Thanks

lianghongzhuo, 2020-02-06:
This is because we use 'DataParallel' while training, this can be solved by using `model = model.module`. And the input point num is also not correct, the correct input point num is 500. In the test file, I use a random numpy array as input point cloud to test the network.
Please pull the new code to solve this issue.

bdenoun, 2020-02-07:
Indeed it works now! Thanks for the fix. It turns out that this line:` points_modify = in_ind_points[ii][np.random.choice(len(in_ind_points[ii]), input_points_num, replace=False)]` always return a  0x3 array, so I guess a fix would be `points_modify = in_ind_points[ii][np.random.choice(len(in_ind_points[ii]), input_points_num, replace=False), :]`

bdenoun, 2020-02-07:
It works indeed! Thanks for the swift fix.

closed 2020-02-07 by bdenoun

trungpham2606, 2020-09-21:
@bdenoun 
Hello, can you send me the pretrained weights of GPD only ?
I cant load the .model file.
So could you please load the .model file, save the model's weights (torch.save(model.state_dict()) only then send me the weights ?
I appreciate your help.