lianghongzhuo/PointNetGPD #20: Grasps too few and unpredicatable

saadehmd, 2020-04-22:
[PointnetGPD.zip](https://github.com/lianghongzhuo/PointNetGPD/files/4515049/PointnetGPD.zip)

I am having trouble generating enough viable grasps repeatedly with  kinect2grasp.py script .
I am using the 3class model and running the inference on an Nvidia Quadro T1000( 4GB). 

I am not sure what i am doing wrong but i am not able to sample enough grasps on the YCB object, even very simple objects like cracker box and mustard bottle. The results are too unpredictable with some objects having a few good grasps, some having not very useful and others having none at all.

I dug through the GPGSamplingPCL script and tried a small change where i changed the dtheta in the grid search to be about normal instead of minor_principalofCurvature. Surprisingly, that generates around 40-50 grasps for almost all ycb objects on average. But most of them are probably colliding with the object. 

I tried removing a few checks here and there in the GPGsampler to pinpoint where all the samples are being rejected but couldn't find much. I am using nonTextured.ply files provided by YCB dataset and converting them to .pcd for ros publishing. The pointcloud looks fine and there's also enough points on it. The final GUI even shows quite a lot of sampled surface points in red color but the actual grasps that get sampled out of those are very few or none.

I am not sure if this description is enough to really tell what's the issue. So, i have attached a few pictures both from 'rotation_about_normal' and 'rotation_about_pc'. I am hoping, maybe you can give some idea, as to what i am doing wrong by just going thru these pictures. 

PS. Thanks in advance :)

lianghongzhuo, 2020-08-03:
You can download the dataset from: https://tams.informatik.uni-hamburg.de/research/datasets/PointNetGPD_grasps_dataset.zip

closed 2020-08-03 by lianghongzhuo

trungpham2606, 2020-09-24:
@saadehmd 
How can you load the 3 class model ? 
It seems to be broken since the author change some file in his workspace so that I cant load it using torch.load now.
Can you send me the weights of the model only ?
I appreciate your help.