lianghongzhuo/PointNetGPD #21: grasp-pointnet installation issues

nico358, 2020-05-15:
Hi @lianghongzhuo,
I experienced some problems when installing this repo:
- The ```meshpy``` package depends on an older ```libboost``` version.  ```boost/numpy.hpp``` from Boost.NumPy is deprecated and has been moved to ```boost/python/numpy.hpp```.  ```boost/python/numeric.hpp``` is also not longer available at (my) version 1.65.
Can you please tell which ```libboost``` version you use or even update the dependencies?
- Modified ```dex-net``` depends on ```Boost.NumPy``` as well and there are numerous other import warnings when I run ```generate-dataset-canny.py```.  I saw that you removed several files from the original repository, eg. ```requirements.txt```. Can I safely ignore the warnings or will cause this problems with the generated dataset? Below I post the relevant output.

Thank you very much!

nic@salocin:~/code$ /usr/bin/python3 /home/nic/code/grasp-pointnet/dex-net/apps/generate-dataset-canny.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
Unable to import meshrender shared library! Rendering will not work. Likely due to missing Boost.Numpy
Boost.Numpy can be installed following the instructions in https://github.com/ndarray/Boost.NumPy
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import Primsense sensor modules! Likely due to missing OpenNI2.
WARNING:root:Failed to import ROS in phoxi_sensor.py. PhoXiSensor functionality unavailable.
WARNING:root:Unable to import generic sensor modules!.
WARNING:root:Failed to import gqcnn! Grasp2D functions will not be available.
WARNING:root:Failed to import OpenRAVE
WARNING:dexnet.grasping.grasp_sampler:Failed to import OpenRAVE
WARNING:dexnet.grasping.grasp_sampler:Failed to import rospy, you can't grasp now.
WARNING:dexnet.api:Failed to import DexNetVisualizer3D, visualization methods will be unavailable
All job done.
a worker of task 002_master_chef_can start
...

lianghongzhuo, 2020-05-16:
We plan to remove some of the dependencies. And right now, you can ignore the warnings, if the code runs. And we will update the `requirement.txt` then.

lianghongzhuo, 2020-05-16:
I hope this commit can let you feel less confused.
https://github.com/lianghongzhuo/PointNetGPD/commit/095866183db1f1180db529306ee1be54601c7515

nico358, 2020-05-16:
I have installed the repository from scratch now. No more include errors. Many thanks for your quick response! 

Currently I'm running the ```generate-dataset-canny.py``` script and the warnings about failed imports popped up but the code runs. I'll try to generate a dataset for ```002_master_chef_can``` object only. I use the preset ```google_512k``` data and process it on my laptop. Last time I aborted the processing because it ran for > 15 hours (due to homeoffice I only use a Intel Core i5-4200M processor with two cores and 4 threads). Can you confirm those computing times for a single object?

closed 2020-05-16 by nico358

lianghongzhuo, 2020-05-16:
Did you enable multiprocessing? 
https://github.com/lianghongzhuo/PointNetGPD/blob/master/dex-net/apps/generate-dataset-canny.py#L157
Be careful not to set this number too large.

nico358, 2020-05-17:
The ```pool_size``` variable is 1 at my installation. I have used the script with its initial values. But the ```do_job``` function releases a lot of threads I guess. The 4 threads at my CPU are continuously at 100% utilization.