lianghongzhuo/PointNetGPD #22: Installation in Anaconda

kaku-KRS, 2020-05-19:
Hi

I am new to programming.
It's suggested to install in a virtual environment, does it mean Anaconda?

If so, I tried install dependencies by : conda install --yes --file requirements.txt , but I got : 

PackagesNotFoundError: The following packages are not available from current channels:

  - autolab-core
  - pyquaternion
  - nearpy
  - autolab-perception
  - trimesh
  - pyhull
  - visualization

I also tried to use pip install, but it seems the packages are installed in my  /.local/lib/python3.6/site-packages, not in the created conda env.

How to solve this?

Thanks

Best Regards

lianghongzhuo, 2020-05-19:
These requirements cannot install through conda, you should use pip instead.
```
pip install -r requirements.txt
```

kaku-KRS, 2020-05-19:
Hi 
I passed pip install, but I got another error in install meshpy :

/home/user/code/grasp-pointnet/meshpy/meshpy/meshrender.cpp:3:10: fatal error: boost/python/numeric.hpp: No such file or directory
 #include "boost/python/numeric.hpp"  

I says I should install libboost 1.58, currently it is 1.65.1.0. However there seems no matching version for Ubuntu 18.04 : https://packages.ubuntu.com/search?keywords=libboost-

Best Regards

lianghongzhuo, 2020-05-19:
Please pull the newest code, I removed this warning.

kaku-KRS, 2020-05-19:
Hi

I am following.

I still got some errors in installing python-pcl

/home/user/.conda/envs/pointnetgpd/compiler_compat/ld: cannot find -lvtkalglib-7.0
/home/user/.conda/envs/pointnetgpd/compiler_compat/ld: cannot find -lvtkChartsCore-7.0
/home/user/.conda/envs/pointnetgpd/compiler_compat/ld: cannot find -lvtkCommonColor-7.0
/home/user/.conda/envs/pointnetgpd/compiler_compat/ld: cannot find -lvtkCommonComputationalGeometry-7.0
/home/user/.conda/envs/pointnetgpd/compiler_compat/ld: cannot find -lvtkCommonCore-7.0
......

I couldn't find the above packages in packages.ubuntu.com.

Thanks

Best Regards

lianghongzhuo, 2020-05-19:
This seems to a version conflict, see https://github.com/strawlab/python-pcl/issues/317

lianghongzhuo, 2020-06-03:
Hi, after some debug, I finally install python-pcl in a condo environment on ubuntu 18,
you should use a python3.6 environment, and install vtk: `conda install vtk=8.1`
Then use my python-pcl fork as a work around.
https://github.com/lianghongzhuo/python-pcl.git

lochinasc1017, 2020-08-18:
I have install python-pcl by `pip install python-pcl`， but i don't know the difference between this method and yours

lianghongzhuo, 2020-08-18:
@lochinasc1017 Their is no difference, only if you can not install `python-pcl` via `pip`,  you might try my workaround.

Guptajakala, 2021-04-07:
Hi,  under python3.6, I'm unable to install autolab_perception due to h5py. How did you solve it? @lianghongzhuo

lianghongzhuo, 2021-04-08:
@Guptajakala Hi, In a clean python 3.6 environments, I can install autolab_perception without any problem. What's the error msg?
Output of my terminal:
```
Successfully built simplegeneric future
Installing collected packages: six, numpy, wcwidth, threadpoolctl, scipy, ruamel.yaml.clib, pytz, python-dateutil, pyparsing, ptyprocess, Pillow, kiwisolver, joblib, ipython-genutils, dill, decorator, cycler, cached-property, traitlets, tifffile, simplegeneric, setproctitle, scikit-learn, ruamel.yaml, pyyaml, PyWavelets, pygments, prompt-toolkit, pickleshare, pexpect, networkx, multiprocess, matplotlib, imageio, h5py, future, colorlog, scikit-video, scikit-image, pyserial, opencv-python, keras, ipython, ffmpeg-python, autolab-core, autolab-perception
Successfully installed Pillow-8.2.0 PyWavelets-1.1.1 autolab-core-0.0.14 autolab-perception-0.0.8 cached-property-1.5.2 colorlog-4.8.0 cycler-0.10.0 decorator-4.4.2 dill-0.3.3 ffmpeg-python-0.2.0 future-0.18.2 h5py-3.1.0 imageio-2.9.0 ipython-5.5.0 ipython-genutils-0.2.0 joblib-1.0.1 keras-2.4.3 kiwisolver-1.3.1 matplotlib-2.2.0 multiprocess-0.70.11.1 networkx-2.5.1 numpy-1.19.5 opencv-python-4.5.1.48 pexpect-4.8.0 pickleshare-0.7.5 prompt-toolkit-1.0.18 ptyprocess-0.7.0 pygments-2.8.1 pyparsing-2.4.7 pyserial-3.5 python-dateutil-2.8.1 pytz-2021.1 pyyaml-5.4.1 ruamel.yaml-0.17.4 ruamel.yaml.clib-0.2.2 scikit-image-0.17.2 scikit-learn-0.24.1 scikit-video-1.1.11 scipy-1.5.4 setproctitle-1.2.2 simplegeneric-0.8.1 six-1.15.0 threadpoolctl-2.1.0 tifffile-2020.9.3 traitlets-4.3.3 wcwidth-0.2.5

```

Guptajakala, 2021-04-08:
Thanks! I recreated a new env and it's solved.

closed 2021-04-16 by lianghongzhuo