lianghongzhuo/PointNetGPD #24: Python version

trungpham2606, 2020-09-08:
Dear @lianghongzhuo ,
Which Python version were you using ?

lianghongzhuo, 2020-09-08:
I am now using python3. Because `numba` is not supporting python2 anymore.

trungpham2606, 2020-09-08:
@lianghongzhuo 
Can you please your ROS version ?
I tried to build ROS melodic from source for Python 3. successfully but unable for moveit till now.

lianghongzhuo, 2020-09-08:
melodic

trungpham2606, 2020-09-08:
@lianghongzhuo 
The dataset you gave link (i found on the issues section ) doesnt have pickle files but npy files only.
Can you check the link again.

lianghongzhuo, 2020-09-08:
That's correct, only npy files

trungpham2606, 2020-09-08:
@lianghongzhuo 
That means we cant visualize the data with read_grasps_from_file.py as you had stated in readme section ?

lianghongzhuo, 2020-09-08:
You need to modify the code a little bit. It's not so hard. change `load pickle` to ` np.load("filename.npy")

trungpham2606, 2020-09-08:
That was exactly what i have done but it seemed something wrong with grasp
score. I will check it again. Tks for your support.

trungpham2606, 2020-09-09:
@lianghongzhuo 
I see in the function show_selected_grasps_with_color(m, ply_name_, title, obj_)
m is grasp_score with shape (6500, 12) loaded from the .npy file you had provided.

m_good = m[m[:, 1] <= 0.4]
m_good = m_good[np.random.choice(len(m_good), size=25, replace=True)]
m_bad = m[m[:, 1] >= 1.8]
m_bad = m_bad[np.random.choice(len(m_bad), size=25, replace=True)]

the maximum value of m[:, 1] is 0.2 therefore m_bad will be empty list. 
Is anything wrong here ?

lianghongzhuo, 2020-09-09:
Hi, please pull the new code, and the new code should fix your problem and visualize the grasps.

trungpham2606, 2020-09-10:
@lianghongzhuo 
Thank you so much for your help !

trungpham2606, 2020-09-10:
@lianghongzhuo 
Sorry for the inconvenience but I can load your provided pretrained model (pointgpd_3class.model) due to the fact that saving the whole model in pickle makes serialized data is bound to the specific classes and the exact directory structure used, so it can break in various ways when used in other projects, or after some serious refactors.

Hence I am not able to load your pretrained model with torch.load(...)

Can you reshare the model link or release the weights of the model only ?

closed 2020-09-17 by trungpham2606