lianghongzhuo/PointNetGPD #26: ICP transform is not correct

trungpham2606, 2020-09-28:
@lianghongzhuo 
I have tested your provided ICP transfrom (from google pointcloud to depth pointcloud)
For 003_cracker_box, the matching result is like below:
[image]

lianghongzhuo, 2020-09-28:
This object is tricky, if you combine multiple point clouds into the scene, you will find that the length of the box in google mesh and rgbd mesh is different. (see the picture blow) So it's hard to say which transform is correct.
[image]

trungpham2606, 2020-09-29:
@lianghongzhuo 
So we should not put it inside the training dataset.
Can you please release the weights again because I had trained the network from scratch but it led me to nothing (the results predicted from pretrained model were not good).
I want to test the potential of the model first before creating new dataset for training.

trungpham2606, 2020-09-29:
@lianghongzhuo 
Some grasp generated is not correct
[image]
[image]

lianghongzhuo, 2020-09-29:
The weight is here:
https://github.com/lianghongzhuo/PointNetGPD/blob/master/data/pointgpd_3class.model

trungpham2606, 2020-09-29:
@lianghongzhuo 
I know it was there but because you saved the whole model to a pkl file so when you change some file or folders in  the project, other people cant load it by using torch.load(r'....'). that why I was asking you to send the weights file only, not the whole model.

lianghongzhuo, 2020-09-29:
What's the error msg when you load the model? I have no problem loading this.

trungpham2606, 2020-09-29:
[image]
here is the error

trungpham2606, 2020-09-29:
When I opened the .model file, 
[image]
I can see the change here.
But rename pointGPD to PointNetGPD doesnt fix the error.
Can you reload the model, then save only the weights (torch.save(model.state_dict(),... ) ?

lianghongzhuo, 2020-09-29:
This looks like a pickle error. See [here](https://github.com/CSAILVision/places365/issues/25#issuecomment-333871990)
Could you try add these codes before loading the model?

trungpham2606, 2020-09-29:
@lianghongzhuo 
Now the error is 
[image]

trungpham2606, 2020-09-29:
I had tested loading the pretrain model in several PC but all of them were not able to load the model.

lianghongzhuo, 2020-09-29:
Please try on a ubuntu 18.04 computer, Windows is not supported.

trungpham2606, 2020-09-29:
@lianghongzhuo 
I also tested on Ubuntu 18.04 PC. But still can not load the model successfully.
I believe that you cant load this model in other PC.

lianghongzhuo, 2020-09-29:
Which code are you testing? I have no problem loading the mode on many computers.

trungpham2606, 2020-09-29:
@lianghongzhuo 
Just simple a test code like this:

import torch
model = torch.load(r''directory to weights')

then the error occurs.

lianghongzhuo, 2020-09-29:
Just run this [file](https://github.com/lianghongzhuo/PointNetGPD/blob/master/PointNetGPD/main_test.py)
And put the error msg here.

trungpham2606, 2020-09-29:
@lianghongzhuo 
So strange that when running that main_test.py I dont catch that error.
I realize that the file I had tested is on other folder that outside the PointNetGPD folder. Maybe that is the problem. 
Thank you so much for spending time on my problem.

closed 2020-09-29 by trungpham2606