lianghongzhuo/PointNetGPD #27: pretrained results

trungpham2606, 2020-09-30:
@lianghongzhuo 
I wrote a script based on your kinect2grasp.py to detect grasp poses from point cloud captured from a camera.
The detection part is very good (I generated around 15 grasps for better visualization):
[image]

But the classification part is very bad that the model classified all the generated grasps as bad grasps.
[image]

Do you have any comment about the results I got ?

lianghongzhuo, 2020-09-30:
Could you share the code and pointcloud you are using? Otherwise, I can't know what's wrong here.

trungpham2606, 2020-09-30:
can I have your mail ? @lianghongzhuo

trungpham2606, 2020-09-30:
@lianghongzhuo 
here is the pointcloud and my python code for testing
https://drive.google.com/drive/folders/1JlKvVFy84muuC2AvBBimDjkmID7k7vi3?usp=sharing

lianghongzhuo, 2020-09-30:
Thank you for the files, I downloaded them. However, I'm busy catching up a deadline, I will check it when I am free.

trungpham2606, 2020-09-30:
@lianghongzhuo 
Thank you. Iam also trying to figure out what is happening when I trained YCB dataset with your pretrained, the loss is quite fluctuating.
I trained 1 object only.

trungpham2606, 2020-10-03:
@lianghongzhuo 
After making several checks, I think the problem comes from the pretrained model.
I also tested the pretrained model in my data and YCB data as well.
The result for my data is above, and here is for YCB:
[image]

Then the model also outputs all generated grasp poses are bad grasp.
I also loaded the model and made a training on YCB dataset, the loss is extremely fluctuating.

trungpham2606, 2020-10-07:
@lianghongzhuo 
Can you send me the old weights : pointgpd_chann3_local.model ?

trungpham2606, 2020-10-14:
@lianghongzhuo 
Can you provide the log file of training session ?

lianghongzhuo, 2020-10-14:
sorry, I don't save the log file. 
I quick looked commit log, https://github.com/lianghongzhuo/PointNetGPD/commit/e2fce0f5dd4a47dfd979e384e562266014d73e41
Seems I only uploaded one model. There is no "old model"

trungpham2606, 2020-10-14:
@lianghongzhuo 
Have you checked my scripts that I had sent to you ?

There's error in PointNetGPD/model/dataset.py, in collect_pc function, you missed the line for recalculate the grasp_bottom_center (grasp_bottom_center = -self.ags.gripper.hand_depth * approach + center).

But after changing it and making some inspections, I still cant work with your pretrained model.

ACaseOfOrange, 2020-11-13:
@trungpham2606 when you run the 3class model, is the 3th element of the output always zero? And the 1st element  of the output always one? The class of the grasp is always 0 or 1 but 2. But some grasps seem to be like good enough.

trungpham2606, 2020-12-24:
@lianghongzhuo 
Had you taken a look at my scripts ?

lianghongzhuo, 2020-12-24:
Sorry for the delay. I will check it today or tomorrow.

trungpham2606, 2020-12-24:
@lianghongzhuo 
oh tks ^^
I also changed the pointnet, and retrained the dataset (eliminate the wrong objects), the results are pretty good.
But I think it'll be better if I can use your pretrained weights.
Hope to hear from you soon !

lianghongzhuo, 2020-12-25:
I checked your code. You have downsampled the input point cloud. But by downsample it . The object point cloud are also down sampled.
The best way is to transform the point cloud into desktop coordinate. Then clip all the point cloud that belong to the table.

trungpham2606, 2021-01-04:
@lianghongzhuo 
Can you test your code on YCB dataset and send me the results ?

lianghongzhuo, 2021-01-04:
The downsample method is not correct, as the table has a lot of points, downsample all the points together will lose some geometric information. You can first crop the points on the table. By doing that you can keep most geometric information of the object.

trungpham2606, 2021-01-04:
@lianghongzhuo 
So what happened with the YCB objects ? without any points on table. The results are the same with no good poses ?

lianghongzhuo, 2021-01-04:
I currently cannot access to the robot in the lab due to the lockdown. When I come back to the office, I can record some point cloud that I use in the experiment. Maybe you can compare the difference. The networks weights work well in my experiment setup.

trungpham2606, 2021-01-06:
@lianghongzhuo 
I think you dont have to test with robot, just run my script I sent to you and see what's happening. The script I wrote for testing YCB object so that this model must work well on it.

lianghongzhuo, 2021-01-06:
as I said, the problem in your script is that you downsampling all the point cloud in the ply file. the correct way to do downsampling is to first crop the table points and then downsample the object point cloud only.

trungpham2606, 2021-01-06:
@lianghongzhuo 
Opps the scripts I sent you was not fixed.
Iam gonna send you the script for testing on YCB object.
`import numpy as np
from voxelgrid import VoxelGrid
import pcl
from autolab_core import YamlConfig
from dexnet.grasping import RobotGripper
from dexnet.grasping import GpgGraspSamplerPcl
import os
import glob
import time
import torch
from scipy.stats import mode
import open3d as o3d
# import multiprocessing as mp
# from PointNetGPD.model.pointnet import PointNetCls
from mayavi import mlab
from pyntcloud import PyntCloud
from numba import jit
from model.gpd import *
from main_test import model

# global config:
yaml_config = YamlConfig(r'G:\trung_grasps\PointNetGPD\dex-net\test\config.yaml')
gripper_name = 'robotiq_85'
gripper = RobotGripper.load(gripper_name, r'G:\trung_grasps\PointNetGPD\dex-net\data\grippers')
ags = GpgGraspSamplerPcl(gripper, yaml_config)
value_fc = 0.4  # no use, set a random number
num_grasps = 10
num_workers = 8
max_num_samples = 100

minimal_points_send_to_point_net = 20
marker_life_time = 8

show_bad_grasp = False
save_grasp_related_file = False

show_final_grasp = True
single_obj_testing = False  # if True, it will wait for input before get pointcloud
check_pcd_grasp_points = False

def get_voxel_fun(points_, n):
    get_voxel = VoxelGrid(points_, n_x=n, n_y=n, n_z=n)
    get_voxel.compute()
    points_voxel_ = get_voxel.voxel_centers[get_voxel.voxel_n]
    points_voxel_ = np.unique(points_voxel_, axis=0)
    return points_voxel_

def get_voxel_fun_win(points, n):
    points_ = PyntCloud.from_instance('open3d', points)
    voxel_id = points_.add_structure('voxelgrid', n_x= n, n_y= n, n_z= n)
    get_voxel = points_.structures[voxel_id]
    points_voxel_ = get_voxel.voxel_centers[get_voxel.voxel_n]
    points_voxel_ = np.unique(points_voxel_, axis=0)
    return points_voxel_

def remove_table_points(points_voxel_, vis=True):
    xy_unique = np.unique(points_voxel_[:, 0:2], axis=0)
    new_points_voxel_ = points_voxel_
    pre_del = np.zeros([1])
    for i in range(len(xy_unique)):
        tmp = []
        for j in range(len(points_voxel_)):
            if np.array_equal(points_voxel_[j, 0:2], xy_unique[i]):
                tmp.append(j)
        if len(tmp) < 3:
            tmp = np.array(tmp)
            pre_del = np.hstack([pre_del, tmp])
[... 291 lines cut ...]
def detect_network(model_, local_pc):
    local_pc = local_pc.T
    local_pc = local_pc[np.newaxis, ...]
    local_pc = torch.FloatTensor(local_pc)
    local_pc = local_pc.cuda()

    output, _ = model_(local_pc)  # N*C
    output = output.softmax(1)
    pred = output.data.max(1, keepdim=True)[1]
    output = output.cpu()
    return pred[0], output.data.numpy()

if __name__ == '__main__':
    input_points_num = 500

    obj_path = r'G:\grasp_dataset\ycb_mesh\004_sugar_box\google_16k\nontextured.ply'
    slm = o3d.io.read_point_cloud(obj_path)
    points = np.array(slm.points)

    m = np.max(points[:, 2]) + 1
    # points[:, 2] = 0.1 + points[:, 2]
    points_reduce = points if len(points) < 50000 else Resampler(points, 30000)

    real_grasp, points, normals_cal = cal_grasp(points_reduce)
`

trungpham2606, 2021-01-06:
@lianghongzhuo 
The result is as below 
[image]
[image]

trungpham2606, 2021-01-14:
@lianghongzhuo 
Excuse me but have you tested my script. Iam looking forward to your help.

trungpham2606, 2021-01-21:
@lianghongzhuo

lianghongzhuo, 2021-01-21:
Sorry for the late reply. 
This model is not trained on the ycb google point cloud. The model is trained on the Raw RGBD data.
For example this  [LINK](http://ycb-benchmarks.s3-website-us-east-1.amazonaws.com/data/berkeley/003_cracker_box/003_cracker_box_berkeley_rgbd.tgz).
If you test on the ycb google point cloud (the laser scan) data, it might fail.
If you have a depth camera (kinect or realsense for example) you can try our model again to see the difference. I will also upload some example point cloud in the future.

trungpham2606, 2021-01-21:
@lianghongzhuo 
Sorry but what is the LINK that you had provided ? When I clicked on it, nothing happened.
Can you double check the LINK ?

trungpham2606, 2021-01-21:
@lianghongzhuo 
We still have to convert the RGBdepth image (captured from the camera) to point cloud, so what is the difference here ? 
Can you make it clear for me ?

lianghongzhuo, 2021-01-21:
http://ycb-benchmarks.s3-website-us-east-1.amazonaws.com/data/berkeley/003_cracker_box/003_cracker_box_berkeley_rgbd.tgz
the link is the download link for ycb rgbd data

trungpham2606, 2021-01-21:
@lianghongzhuo 
Thank you for the link.
1 more problem is that with boxes, I always meet this failure:
[image]
it should be bad pose instead of good pose.

trungpham2606, 2021-01-25:
@lianghongzhuo 
If I want to use different gripper, do I have to re-create the dataset according to my gripper's config ?

trungpham2606, 2021-02-17:
@lianghongzhuo

Twilight89, 2021-03-09:
@trungpham2606 Hi~ I want to ask that in the kinect2grasp.py, 
        kinect_data = rospy.wait_for_message("table_top_points", PointCloud2)
the kinect_data is based on kinect coordinate, right? So I can directly use /kinect_V2/depth/points topic published by kinect? 
And in
        real_grasp, points, normals_cal = cal_grasp(kinect_data, cam_pos)
the points here is also based on kinect coordinate(I checked the data and in my production, it's based on kinect coordinate)? I am confused because I send cam_pos in, hope to get points based on table_top coordinate.
I don't know why, leading to my result like follow:(I use rviz, failing to import mayavi )
Hope for you reply, thanks a lot!
[image]

trungpham2606, 2021-03-09:
@Twilight89 
According to author's code, the objects point cloud are on world coordinate. I didnt use ROS for visualizing but using real camera instead.

Twilight89, 2021-03-09:
@trungpham2606 Thanks a lot for your reply. So points in 
real_grasp, points, normals_cal = cal_grasp(kinect_data, cam_pos)
are on world coordinate? But kinect_data is on camera coordinate, do I need to tranform kinect_data(camera_frame) to points(world_frame). 
I used to think points are on world coordinate, but I checked the data in my repo, they are on camera coordinate. I thought I have sent cam_pos in so the tranformation was done.
Briefly, now I should get points in world frame(now they are camera frame), is that your meaning?
Hope for reply~

trungpham2606, 2021-03-09:
@Twilight89 
Yes, you should get points in world frame.

Twilight89, 2021-03-09:
@trungpham2606 thanks again~
I want to know in your repo, is kinect_data in camera frame? because in this line
   cam_pos, _ = listener.lookupTransform('/table_top', '/kinect2_ir_optical_frame', rospy.Time(0)) 
we can get cam_pos, and send in 
   real_grasp, points, normals_cal = cal_grasp(kinect_data, cam_pos)
I used to think the tranformation was done due to I sent in cam_pos. If not, why we should get cam_pos?
Hoping for reply

trungpham2606, 2021-03-09:
@Twilight89 
Sorry but we didnt run on ROS, we test on real camera(realsense) and robot(UR5) directly. In our experiment, the pointcloud and cam_pos are both on world coordinates.

Twilight89, 2021-03-09:
@trungpham2606 
Sorry I am a little confused. Did you change something about 'data transform' after git this code. I think there is no difference between simulation and reality. 
In my simulation repo, the cam_pos is on world coordinate, too. And I think in real camera, the directly data it got is on camera coordinate, too.
If this, I just found my points are on carema coordinate, which is unnormal, so did you do any transformation to change points to world frame?

Twilight89, 2021-03-09:
@trungpham2606 
Thanks a lot for your patience, I'll try it later. Hope your problem can be solved soon.

trungpham2606, 2021-03-11:
@Twilight89 
Thank you, my problem has been solved. Iam gonna close this issue right now !

closed 2021-03-11 by trungpham2606