lianghongzhuo/PointNetGPD #29: problem about params.json

Ctiger96, 2020-10-20:
Hi, I'm generating my own dataset, but some params in params.json make me confused.

1. What does the param "finger_radius" mean? I can't find the definition in gpg. And what is the difference between "finger_width" and "real_finger_width"? Does the "hand_height" mean the height of the finger?
2. The robot hand that I'm using is RG2, and its depth is changing with the width. For example, when RG2 is fully open, the width is 10.5cm and the depth ( from fingertip to the base ) is 4.5cm at the moment. But when it is closed, the depth can be 8.5cm, much bigger than the former. I have no idea about setting the param to the situation like this.

Would you please give me some suggestions about my problems ? Thx!

trungpham2606, 2020-11-07:
@Ctiger96 
you have to define these parameters only:
finger_width = 
hand_outer_diameter = 
hand_depth = 
hand_height = 
init_bite = 
Ignore these others.

lianghongzhuo, 2020-12-23:
Please see the readme, I uploaded a picture to illustrate the meaning of these parameters.

lianghongzhuo, 2020-12-31:
Close in  favour of 6455e0749aec0f70a4a46b95305c93889c5057f1

closed 2020-12-31 by lianghongzhuo