lianghongzhuo/PointNetGPD #30: do I need to run other ROS nodes separately besides this program

Alex-czh, 2020-11-11:
Hello, do I need to run other ROS nodes separately besides this program?
What does it have to do to directly control the real camera and the manipulator arm?
thanks.

youkenhou, 2020-11-17:
The script `kinect2grasp.py` only uses the pointcloud to generate potential grasps and sends all the grasps into network to get good ones. The good grasps are published as a ros topic. So another node to read the grasp msg and control the arm is neccessary. However now I'm stuck at `kinect2grasp.py`. If you manage to run it successfully, please tell me how.

lianghongzhuo, 2020-12-23:
The script `kinect2grasp.py` will read point cloud from a ros topic. then evaluate it. There is no arm motion involved. You can try `moveit` if you want to plan an arm motion.

closed 2023-05-16 by lianghongzhuo