lianghongzhuo/PointNetGPD #32: Problem with kinect2grasp.py

youkenhou, 2020-11-17:
@lianghongzhuo Firstly thank you for your great paper and code!
When I tried to run kinect2grasp.py, I noticed that there is no defination of `table_top`. So I fixed the value of `cam_top` and used the pointcloud of kinect2 directly to let the procedure continue.
However, I encountered an error just in front of the network, which is 
```bash
Caculated padded input size per channel: (0). Kernel size :(1). Kernel size can't be greater than actual input size
```
The log showed that several graps were actually sent into the network. May you help me figure out the reason why the error occurred?
Thanks in advance.

lianghongzhuo, 2020-12-23:
sorry for the late reply. have you fixed the issue? could you print the shape of your input?

gachiemchiep, 2020-12-29:
Hello @lianghongzhuo 
I met the same error.  

```bash
Traceback (most recent call last):
  File "kinect2grasp.py", line 522, in <module>
    if_good_grasp, grasp_score_tmp = test_network(model.eval(), points_modify)
  File "/home/jil/workspace/2021_grasping/PointNetGPD/dex-net/apps/main_test.py", line 68, in test_network
    output, _ = model_(local_pc)  # N*C
  File "/home/jil/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "../../PointNetGPD/model/pointnet.py", line 186, in forward
    x, trans = self.feat(x)
  File "/home/jil/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "../../PointNetGPD/model/pointnet.py", line 137, in forward
    trans = self.stn(x)
  File "/home/jil/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "../../PointNetGPD/model/pointnet.py", line 30, in forward
    x = F.relu(self.bn1(self.conv1(x)))
  File "/home/jil/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/jil/.local/lib/python2.7/site-packages/torch/nn/modules/conv.py", line 202, in forward
    self.padding, self.dilation, self.groups)
RuntimeError: Calculated padded input size per channel: (0). Kernel size: (1). Kernel size can't be greater than actual input size

```

Here is the full log. 

[PointNetGPD.log](https://github.com/lianghongzhuo/PointNetGPD/files/5750040/PointNetGPD.log)

We don't have kinect sensor so we use realsense instead. 
Since the realsense can output data in PointCloud2 format, we think realsense is fine. 

The size of cloud is : 

```bash
[INFO] [1609230294.825651]: the voxel has 3802 points, we want get 270104 points
Cloud size: size:3802 height: 1 width:3802
```

lianghongzhuo, 2020-12-29:
Could you try run this script? This script will feed some random data to the network for testing.
https://github.com/lianghongzhuo/PointNetGPD/blob/master/PointNetGPD/main_test.py

gachiemchiep, 2020-12-29:
@lianghongzhuo 

I re-defined some path for model loading only. 
Here is the output

```bash
(base) jil@tr_ws:~/workspace/2021_grasping/PointNetGPD/dex-net/apps$  python main_test.py
Using default model file
/home/jil/.local/lib/python2.7/site-packages/torch/serialization.py:593: SourceChangeWarning: source code of class 'torch.nn.parallel.data_parallel.DataParallel' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/jil/.local/lib/python2.7/site-packages/torch/serialization.py:562: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if original_source != current_source:
/home/jil/.local/lib/python2.7/site-packages/torch/serialization.py:593: SourceChangeWarning: source code of class 'torch.nn.modules.conv.Conv1d' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/jil/.local/lib/python2.7/site-packages/torch/serialization.py:593: SourceChangeWarning: source code of class 'torch.nn.modules.pooling.MaxPool1d' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/jil/.local/lib/python2.7/site-packages/torch/serialization.py:593: SourceChangeWarning: source code of class 'torch.nn.modules.linear.Linear' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/jil/.local/lib/python2.7/site-packages/torch/serialization.py:593: SourceChangeWarning: source code of class 'torch.nn.modules.activation.ReLU' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/jil/.local/lib/python2.7/site-packages/torch/serialization.py:593: SourceChangeWarning: source code of class 'torch.nn.modules.batchnorm.BatchNorm1d' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
load model ../../data/pointgpd_3class.model
model structure : DataParallel(
  (module): PointNetCls(
    (feat): PointNetfeat(
      (stn): STN3d(
        (conv1): Conv1d(3, 64, kernel_size=(1,), stride=(1,))
        (conv2): Conv1d(64, 128, kernel_size=(1,), stride=(1,))
        (conv3): Conv1d(128, 1024, kernel_size=(1,), stride=(1,))
        (mp1): MaxPool1d(kernel_size=500, stride=500, padding=0, dilation=1, ceil_mode=False)
        (fc1): Linear(in_features=1024, out_features=512, bias=True)
        (fc2): Linear(in_features=512, out_features=256, bias=True)
        (fc3): Linear(in_features=256, out_features=9, bias=True)
        (relu): ReLU()
        (bn1): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (bn2): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (bn3): BatchNorm1d(1024, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (bn4): BatchNorm1d(512, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (bn5): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      )
      (conv1): Conv1d(3, 64, kernel_size=(1,), stride=(1,))
      (conv2): Conv1d(64, 128, kernel_size=(1,), stride=(1,))
      (conv3): Conv1d(128, 1024, kernel_size=(1,), stride=(1,))
      (bn1): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (bn2): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (bn3): BatchNorm1d(1024, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (mp1): MaxPool1d(kernel_size=500, stride=500, padding=0, dilation=1, ceil_mode=False)
    )
    (fc1): Linear(in_features=1024, out_features=512, bias=True)
    (fc2): Linear(in_features=512, out_features=256, bias=True)
    (fc3): Linear(in_features=256, out_features=3, bias=True)
    (bn1): BatchNorm1d(512, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
    (bn2): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
    (relu): ReLU()
  )
)
voting:  [tensor([0]), tensor([0]), tensor([0]), tensor([0]), tensor([0]), tensor([0]), tensor([0]), tensor([0]), tensor([0]), tensor([0])]
Test result: 0

```

lianghongzhuo, 2020-12-29:
then the model load successfully. maybe you can check if your input is the same as this test script?

gachiemchiep, 2020-12-29:
@lianghongzhuo 

I see this part of code inside test script. Does this mean your network only accept point-cloud with size=500

```bash
def main():
    repeat = 10
    num_point = 500
    model.eval()
    torch.set_grad_enabled(False)

    # load pc(should be in local gripper coordinate)
    # local_pc: (N, 3)
    # local_pc = np.load("test.npy")
    local_pc = np.random.random([500, 3])  # test only
```

My input is different from the test script 

```bash 
[INFO] [1609230294.825651]: the voxel has 3802 points, we want get 270104 points
Cloud size: size:3802 height: 1 width:3802
```

lianghongzhuo, 2020-12-29:
yes, please downsample the input to 500

gachiemchiep, 2020-12-29:
@lianghongzhuo  
Sorry for a novice question but do you know which API should I use to downsample pointcloud data into 500 points. 
https://python-pcl-fork.readthedocs.io/en/rc_patches4/tutorial/filtering.html 

And should  the sampling be added after these lines : 

```bash
def cal_grasp(msg, cam_pos_):
    points_ = pointclouds.pointcloud2_to_xyz_array(msg)
    points_ = points_.astype(np.float32)
    remove_white = False
    if remove_white:
        points_ = remove_white_pixel(msg, points_, vis=True)
    # begin voxel points
    n = n_voxel  # parameter related to voxel method
    # gpg improvements, highlights: flexible n parameter for voxelizing.
    points_voxel_ = get_voxel_fun(points_, n)
    if len(points_) < 2000:  # should be a parameter
        while len(points_voxel_) < len(points_)-15:
            points_voxel_ = get_voxel_fun(points_, n)
            n = n + 100
            rospy.loginfo("the voxel has {} points, we want get {} points".format(len(points_voxel_), len(points_)))

    rospy.loginfo("the voxel has {} points, we want get {} points".format(len(points_voxel_), len(points_)))
    points_ = points_voxel_
    remove_points = False
    if remove_points:
        points_ = remove_table_points(points_, vis=True)
    point_cloud = pcl.PointCloud(points_)
    norm = point_cloud.make_NormalEstimation()
    norm.set_KSearch(30)  # critical parameter when calculating the norms
    normals = norm.compute()
    surface_normal = normals.to_array()
    surface_normal = surface_normal[:, 0:3]
    vector_p2cam = cam_pos_ - points_
    vector_p2cam = vector_p2cam / np.linalg.norm(vector_p2cam, axis=1).reshape(-1, 1)
    tmp = np.dot(vector_p2cam, surface_normal.T).diagonal()
    angel = np.arccos(np.clip(tmp, -1.0, 1.0))
    wrong_dir_norm = np.where(angel > np.pi * 0.5)[0]
    tmp = np.ones([len(angel), 3])
    tmp[wrong_dir_norm, :] = -1
    surface_normal = surface_normal * tmp
    select_point_above_table = 0.010
    #  modify of gpg: make it as a parameter. avoid select points near the table.
    points_for_sample = points_[np.where(points_[:, 2] > select_point_above_table)[0]]
```

lianghongzhuo, 2020-12-29:
I used voxelization to downsample the points: https://github.com/lianghongzhuo/PointNetGPD/blob/master/dex-net/apps/kinect2grasp.py#L143-L145
`n_voxel` was set to 500

gachiemchiep, 2020-12-29:
@lianghongzhuo Thanks for the code. 
I will run the code again soon and let you know the result

gachiemchiep, 2021-01-04:
@lianghongzhuo 
We run the kinect2grasp.py and notify that inside **def cal_grasp(msg, cam_pos_):** method, the input pointcloud is downsampled once by **points_voxel_ = get_voxel_fun(points_, n)**. But unfortunately, somehow the downsampled pointcloud size is still large than 500. We tried to run the  **points_voxel_ = get_voxel_fun(points_, n)** for 2nd time, but the downsampled pointcloud couldn't be downsampled into smaller pointcloud. So the same error occured. 

```bash
def cal_grasp(msg, cam_pos_):
    points_ = pointclouds.pointcloud2_to_xyz_array(msg)
    points_ = points_.astype(np.float32)
    remove_white = False
    if remove_white:
        points_ = remove_white_pixel(msg, points_, vis=True)
    # begin voxel points
    n = n_voxel  # parameter related to voxel method
    # gpg improvements, highlights: flexible n parameter for voxelizing.
    points_voxel_ = get_voxel_fun(points_, n)
    rospy.loginfo("Convert {} points into {} points".format(len(points_), len(points_voxel_)))
    if len(points_) < 2000:  # should be a parameter
        while len(points_voxel_) < len(points_)-15:
            points_voxel_ = get_voxel_fun(points_, n)
            n = n + 100
            rospy.loginfo("the voxel has {} points, we want get {} points".format(len(points_voxel_), len(points_)))

    rospy.loginfo("the voxel has {} points, we want get {} points".format(len(points_voxel_), len(points_)))
    points_ = points_voxel_
    remove_points = False
    if remove_points:
        points_ = remove_table_points(points_, vis=True)

    points_voxel_ = get_voxel_fun(points_, 500)
    rospy.loginfo("Convert {} points into {} points".format(len(points_), len(points_voxel_)))
    points_ = points_voxel_
```

```bash
[INFO] [1609727671.185379]: getting transform from camera_1 (realsense) to table top
[INFO] [1609727675.199613]: got transform complete
[INFO] [1609727675.203337]: Robot is at home, safely catching point cloud data.
[INFO] [1609727675.205075]: rospy is waiting for message: /camera_1/depth_registered/points
[INFO] [1609727678.655435]: get_voxel_fun: points_voxel 4479 points
[INFO] [1609727678.656680]: get_voxel_fun: points 269939 points
[INFO] [1609727678.662804]: Convert 269939 points into 4479 points
[INFO] [1609727678.665078]: the voxel has 4479 points, we want get 269939 points
[INFO] [1609727681.542719]: get_voxel_fun: points_voxel 4479 points
[INFO] [1609727681.543105]: get_voxel_fun: points 4479 points
[INFO] [1609727681.546632]: Convert 4479 points into 4479 points
[INFO] [1609727681.651791]: pointcloud : size=13437, height=4479, width=3
[INFO] [1609727681.652503]: Begin cal grasps using single thread, slow!

```
Do you have any idea about this problem.

lianghongzhuo, 2021-01-04:
Your input point cloud has  269939 points. Maybe this point cloud contains the table? You can first crop the points belong to the table then do the downsample.

qiushu-chen, 2021-02-25:
@gachiemchiep
Hi bro, I meet a problem similar to yours. Since  we donot have the Kinect2 sensor, we tend to use Zed2 camera or Azure Kinect to get the Pointcloud message. However, in kinect2grasp.py, the TF listener uses '/kinect2_ir_optical_frame' to do the transform between the two coordinate systems. Is the transformation necessary for this project? And is the frame a .jpg file or something else? Are there any substitutes for this frame when you are using Realsense, or you just leave it alone?

gachiemchiep, 2021-02-27:
@qiushu-chen 

The loop here shows that you need the transformation before going further.
So you need the transformation. 
```bash
    get_transform = False
    while not get_transform:
        try:
            cam_pos, _ = listener.lookupTransform('/table_top', '/kinect2_ir_optical_frame', rospy.Time(0))
            get_transform = True
            rospy.loginfo("got transform complete")
        except (tf.LookupException, tf.ConnectivityException, tf.ExtrapolationException):
            continue
```

When using Realsensen, I change the name of tf topic in my own setting. 
If you don't have your own environment, you can create a virtual one using ros launch file using *static_transform_publisher*. Like this 

```bash 
<?xml version="1.0"?>
<!-- three camera demo -->
<launch>
    <group ns="$(arg camera_2)">
        <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
        </include>
    </group>
    <node pkg="tf" type="static_transform_publisher" name="camera_2_link"
        args=" 0.21577865 -0.33410704  1.02983987 -0.70355683  0.71061717  0.00165881  0.00531743 world camera_2_color_optical_frame 100"/>
</launch>
```

qiushu-chen, 2021-03-01:
@gachiemchiep @lianghongzhuo
Thanks a lot for your kind advice! Since we don't have our own environment, we will try to build it up using the ros launch file. 
However, I am still not so sure about the relationships between those coordinate systems. Does the link table_top mean the unified local gripper coordinate or something else? How is the table_top coordinate defined?

qiushu-chen, 2021-03-11:
@gachiemchiep @lianghongzhuo
When using the kinect2grasp.py file, we got this feedback:
`[INFO] [1615449004.953655]: Grasp sampler finish, generated 14 grasps.
[14, 16, 17, 12, 13, 13, 11, 11, 11, 13, 11, 14, 11, 21]
[INFO] [1615449004.996441]: Begin send grasp into pointnet, cal grasp score
Traceback (most recent call last):
  File "kinect2grasp.py", line 525, in <module>
    if_good_grasp, grasp_score_tmp = test_network(model.eval(), points_modify)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/main_test.py", line 66, in test_network
    output, _ = model_(local_pc)  # N*C
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/model/pointnet.py", line 186, in forward
    x, trans = self.feat(x)
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/model/pointnet.py", line 137, in forward
    trans = self.stn(x)
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/model/pointnet.py", line 30, in forward
    x = F.relu(self.bn1(self.conv1(x)))
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/conv.py", line 202, in forward
    self.padding, self.dilation, self.groups)
RuntimeError: Calculated padded input size per channel: (0). Kernel size: (1). Kernel size can't be greater than actual input size`
We have already done the downsampling, and broadcasted the tf transforming, so we did not meet the problems like "the voxel has 4479 points, we want get 269939 points". However here comes the problem RuntimeError. Did you meet such kind of problem?
Another question is about the tf transforming. Do we still need to transform the points in the Pointcloud2 message to the coordinate of the foundation of the robot before we broadcast the topic? Or this step will be completed by the tf transformer in this program?
`    while not get_transform:
        try:
            #cam_pos, _ = listener.lookupTransform('/table_top', '/kinect2_ir_optical_frame', rospy.Time(0))
            cam_pos, _ = listener.lookupTransform('/base_link', '/camera_link', rospy.Time(0))
            get_transform = True
            rospy.loginfo("got transform complete")
        except (tf.LookupException, tf.ConnectivityException, tf.ExtrapolationException):
            continue`

qiushu-chen, 2021-03-11:
@gachiemchiep @lianghongzhuo
When using the kinect2grasp.py file, we got this feedback:

> [INFO] [1615449004.953655]: Grasp sampler finish, generated 14 grasps.
[14, 16, 17, 12, 13, 13, 11, 11, 11, 13, 11, 14, 11, 21]
[INFO] [1615449004.996441]: Begin send grasp into pointnet, cal grasp score
Traceback (most recent call last):
  File "kinect2grasp.py", line 525, in <module>
    if_good_grasp, grasp_score_tmp = test_network(model.eval(), points_modify)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/main_test.py", line 66, in test_network
    output, _ = model_(local_pc)  # N*C
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/model/pointnet.py", line 186, in forward
    x, trans = self.feat(x)
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/model/pointnet.py", line 137, in forward
    trans = self.stn(x)
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/projects/PointNetGPD/PointNetGPD/model/pointnet.py", line 30, in forward
    x = F.relu(self.bn1(self.conv1(x)))
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cqsfdu/anaconda3/envs/dextest/lib/python2.7/site-packages/torch/nn/modules/conv.py", line 202, in forward
    self.padding, self.dilation, self.groups)
RuntimeError: Calculated padded input size per channel: (0). Kernel size: (1). Kernel size can't be greater than actual input size

We have already done the downsampling, and broadcasted the tf transforming, so we did not meet the problems like "the voxel has 4479 points, we want get 269939 points". However here comes the problem RuntimeError. Did you meet such kind of problem?
Another question is about the tf transforming. Do we still need to transform the points in the Pointcloud2 message to the coordinate of the foundation of the robot before we broadcast the topic? Or this step will be completed by the tf transformer in this program?

>     while not get_transform:
        try:
            #cam_pos, _ = listener.lookupTransform('/table_top', '/kinect2_ir_optical_frame', rospy.Time(0))
            cam_pos, _ = listener.lookupTransform('/base_link', '/camera_link', rospy.Time(0))
            get_transform = True
            rospy.loginfo("got transform complete")
        except (tf.LookupException, tf.ConnectivityException, tf.ExtrapolationException):
            continue

lianghongzhuo, 2021-03-11:
Please only put points on the table as the input of this file. otherwise downsample all the points from your camera will lose a lot of information about the objects on the table. 
and also run this code with `--cuda`

qiushu-chen, 2021-03-15:
Dr. Liang:
Thanks a lot for your advice! In fact, after that we tried to downsample the pointcloud and send it to the python file again, without getting the error information we mentioned before. However, we finally got such an output of the grasps shown in output[7]:
As the grasps returned by the program were all integers, while the grasps in your grasp datasets are like the output[9], we are not sure about the meanings of those integers. Also, another question is about the tf transforming. Do we still need to transform the points in the Pointcloud2 message to the coordinate of the foundation of the robot before we broadcast the topic? Or this step will be completed by the tf transformer in this program?
Looking forward to your kind advice! And best wishes.
Yours Sincerely

qiushu-chen, 2021-03-15:
Dr. Liang @lianghongzhuo:
Thanks a lot for your advice! In fact, after that we tried to downsample the pointcloud and send it to the python file again, without getting the error information we mentioned before. However, we finally got such an output of the grasps:
`[17 11 11 11 11 11 11 11 13 16 12 14 11 12 12]`
As the grasps returned by the program were all integers, while the grasps in your grasp datasets are like the output:
`[[ 0.00762354 -0.01083387  0.01377152 ...  0.          1.6  0.27384749]
...
[-0.00986061 -0.00513373  0.15122774 ...  0.          0.5  0.41621156]]`
, we are not sure about the meanings of those integers. Also, another question is about the tf transforming. Do we still need to transform the points in the Pointcloud2 message to the coordinate of the foundation of the robot before we broadcast the topic? Or this step will be completed by the tf transformer in this program?
Looking forward to your kind advice! And best wishes.

lianghongzhuo, 2021-03-15:
The grasp dataset is only for training. To run the NN, you don't need to load that dataset.
Pointcloud2 and tf message is for real robot experiments. You don't need that if you do not use ros and real robot.

cxtcatherine, 2021-06-22:
Sorry to request a question about the python3 and ros. How can they work together in kinect2grasp.py?

lianghongzhuo, 2021-06-22:
@cxtcatherine  hi, what's the error msg when you run `kinect2grasp.py` in python3?

cxtcatherine, 2021-06-23:
> @cxtcatherine hi, what's the error msg when you run `kinect2grasp.py` in python3?

Traceback (most recent call last):
  File "kinect2grasp.py", line 20, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)

lianghongzhuo, 2021-06-23:
I already removed the import of tf at c61ab1111d08007ecb41972142acdb67ed6a496d
Please pull the new code. And hard code `cam_pos` at https://github.com/lianghongzhuo/PointNetGPD/blob/master/dex-net/apps/kinect2grasp.py#L408

lianghongzhuo, 2021-06-24:
Close this as it is addressed in the new commit, if you have further questions, feel free to open a new issues.

closed 2021-06-24 by lianghongzhuo