lianghongzhuo/PointNetGPD #33: Problems about gripper

Jay-9912, 2020-11-23:
I notice that the gripper configuration file named robotiq_85.  Does it mean that the configuration is for robotiq 2F 85 gripper? Since I am using robotiq 2F 85, I want to know whether I need to modify the configuration. Thx!

lianghongzhuo, 2020-12-23:
I will upload a figure to illustrate the meaning of these parameters. Please see the readme.

Jay-9912, 2020-12-26:
Thanks. Another question I'd like to ask is that in kinect2grasp.py, what's the meaning of cam_pos in line 443? Does it mean the position of camera in world coordinate? And is the points_ in line 137 also in world coordinate?

Jay-9912, 2020-12-26:
> I uploaded a figure to illustrate the meaning of these parameters. Please see the readme.
> [6455e07](https://github.com/lianghongzhuo/PointNetGPD/commit/6455e0749aec0f70a4a46b95305c93889c5057f1)

Thanks. Another question I'd like to ask is that in kinect2grasp.py, what's the meaning of cam_pos in line 443? Does it mean the position of camera in world coordinate? And is the points_ in line 137 also in world coordinate?

lianghongzhuo, 2020-12-26:
Here I use cam_pos to calculate the object norm direction. This is crucial for the gripper approach direction. Points_ are at `table_top` frame. this is handy for remove points that belongs the table.