lianghongzhuo/PointNetGPD #35: How to perform a fresh installation in a docker container

JordiMM, 2021-02-06:
I'm trying to use the PointNetGPD in a docker container to have an isolated installation.

What is the required environment to have PointNetGDP working?
From the issues sections I determined that the best safe path is to use ubuntu 18.04 with ros melodic but until now I couldn't manage to obtain a working environment. After following the instructions, I have problems when I try to use the trained network.

I explain a bit the steps I performed:

I pulled a pytorch docker image (pytorch/pytorch:1.7.1-cuda11.0-cudnn8-runtime) and I installed the code in the $HOME/code path, followed the install requirement instructions and downloaded the Generated Grasp Dataset. Afterwards I installed ros-melodic.
At this point, as I don't want to generate a new database I jumped to the Using the trained network instructions and executed "python get_ur5_robot_state.py".
It reported me that the module rospy doesn't exist.
From this point I install a bunch of python and ros packages via pip and apt and the current error I have is:
```bash
root@534bf88c2be0:~/code/PointNetGPD/dex-net/apps# python get_ur5_robot_state.py
Traceback (most recent call last):
  File "get_ur5_robot_state.py", line 29, in <module>
    group = moveit_commander.MoveGroupCommander("arm")
AttributeError: module 'moveit_commander' has no attribute 'MoveGroupCommander'
```

And I'm not able to move forward.

I also have problems regarding kinect2grasp.py.
I think the error is about I lack some python libraries:
```bash
root@534bf88c2be0:~/code/PointNetGPD/dex-net/apps# python kinect2grasp.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
2021-02-06 18:57:55.095252: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Failed to import ROS in Kinect2_sensor.py. Kinect will not be able to be used in bridged mode
WARNING:root:Unable to import Primsense sensor modules! Likely due to missing OpenNI2.
WARNING:root:Unable to import pyrealsense2.
WARNING:root:Failed to import ROS in ensenso_sensor.py. ROS functionality not available
WARNING:root:Failed to import ROS in phoxi_sensor.py. PhoXiSensor functionality unavailable.
WARNING:root:Unable to import generic sensor modules!.
/opt/conda/lib/python3.8/site-packages/traits/etsconfig/etsconfig.py:412: UserWarning: Environment variable "HOME" not set, setting home directory to /tmp
  warn(
Please install grasp msgs from https://github.com/TAMS-Group/gpd_grasp_msgs in your ROS workspace
```

I have a working protonect from libfreenect and I installed pylibfreenect2, but it seems it is not detected (maybe a python version problem?)

Any help in what I can do to be able to have the PointNetGPD working?

lianghongzhuo, 2021-02-06:
These are code that related to ros (for control a real robot), I haven't published ros control code yet.
I plan to do it around the end of March.
Currently,  you can test the network or write your own code to control the robot.

JordiMM, 2021-02-08:
Ok. Thanks for the answer.
I will try to test the network.

closed 2021-02-08 by JordiMM