lianghongzhuo/PointNetGPD #36: About the YCB dataset

qiushu-chen, 2021-02-17:
When using the YCB dataset, we were not sure how can we get the following files: ycb_meshes_google  (YCB dataset)、 ycb_rgbd  (YCB dataset). We downloaded the generated npy files, among with the files in "berkeley_processed" and "google_64k", however it seems that those files are not included. We wander whether there are some other files should be  downloaded, or should we do some other works to generate them from the original files or not? Could you please give us some advice?
Also, would you like to tell us something about your hardware basis when doing this project? We are trying to reappearance this network on a notebook with a Geforce RTX 2060 GPU (whose capacitance is 6G) and an intel Core i5 CPU.    Is it enough for this project?
We are looking forward to your kind advice!

qiushu-chen, 2021-02-17:
@lianghongzhuo 
When using the YCB dataset, we were not sure how can we get the following files: ycb_meshes_google (YCB dataset)、 ycb_rgbd (YCB dataset). We downloaded the generated .npy files, among the files in "berkeley_processed" and "google_64k", however it seems that those files are not included. We wonder whether there are some other files that should be downloaded, or should we do some other works to generate them from the original files or not? Could you please give us some advice?
Also, would you like to tell us something about your hardware basis when doing this project? We are trying to reappearance this network on a notebook with a Geforce RTX 2060 GPU (whose capacitance is 6G) and an Intel Core i5 CPU. Is it enough for this project?
We are looking forward to your kind advice!

qiushu-chen, 2021-02-25:
@lianghongzhuo 
There's another question. Since we donot have the Kinect2 sensor, we tend to use Zed2 camera or Azure Kinect to get the Pointcloud message. However, in kinect2grasp.py, the TF listener uses '/kinect2_ir_optical_frame' to do the transform between the two coordinate systems. Is the transformation necessary for this project? And is the frame a .jpg file or something else?

Josin1, 2021-03-11:
@lianghongzhuo 
Hi, 

> @lianghongzhuo
> When using the YCB dataset, we were not sure how can we get the following files: ycb_meshes_google (YCB dataset)、 ycb_rgbd (YCB dataset). We downloaded the generated .npy files, among [...]

I do have the same doubt.

lianghongzhuo, 2021-03-11:
ycb dataset can be download from: http://ycb-benchmarks.s3-website-us-east-1.amazonaws.com/

lianghongzhuo, 2021-03-11:
or use https://github.com/lianghongzhuo/ycb-tools 
to download. 
run: `python3 download_ycb_dataset.py`

qiushu-chen, 2021-03-20:
Dr. Liang:
We have already got the program work, hoewever, the results are a little strange. Like the following image, we put the pointcloud taken from the vertical view of the objects. In the results, we can only get the grasps from the side directions and the network always show that they are bad grasps. Could you please give us some kind advice?
[image]
[image]

qiushu-chen, 2021-05-06:
Dr.Liang @lianghongzhuo :
Hello! As the main_test file uses the pointgpd_3class.model as the pretrained grasp model, we want to make sure what kinds of objects are trained in this model. Are most of the objects among the YCB dataset included in this model, or only a few objects are trained as an example?

lianghongzhuo, 2021-05-06:
This is the dataset and object name we used for training: 
https://tams.informatik.uni-hamburg.de/research/datasets/PointNetGPD_grasps_dataset.zip

qiushu-chen, 2021-05-06:
Dr. Liang@lianghongzhuo:
Thanks a lot. We have already downloaded this dataset. And do you mean that those objects have already been included in the pre-trained model pointgpd_3class.model?

lianghongzhuo, 2021-05-06:
I'm not sure if we removed some of the objects during the training, as it's a long time ago. But I can confirm most of the objects in our grasp dataset are contained in the model.

xiaopgaxiaopg, 2022-06-22:
你好，可以留个联系方式，交流下这个算法吗，可以付费咨询。

xiaopgaxiaopg, 2022-06-29:
> @lianghongzhuo When using the YCB dataset, we were not sure how can we get the following files: ycb_meshes_google (YCB dataset)、 ycb_rgbd (YCB dataset). We downloaded the generated .npy files, among [...]
你好，可以请教下训练数据集吗，我这边有点疑问

closed 2023-06-08 by lianghongzhuo