lianghongzhuo/PointNetGPD #39: Computing grasp score is different from written in the paper?

Guptajakala, 2021-04-09:
Hi, thanks for the great repo!
In your paper, if the friction coeff is `miu`, then the score should be `1/miu`. But in your implementation, I found you are directly adding `miu` as part of the score.
https://github.com/lianghongzhuo/PointNetGPD/blob/5d3453df192866968d03569eb4da66b0fba8bfd6/PointNetGPD/model/dataset.py#L265
The level_score is basically the friction coeff `miu`. Is my understanding right?

lianghongzhuo, 2021-04-09:
The `level_score = 1/miu`, so the lower the friction coeff, the higher the score

Guptajakala, 2021-04-09:
During grasp training data generation, I saw here the second term of a grasp is value_fc, is it the `miu`?
https://github.com/lianghongzhuo/PointNetGPD/blob/5d3453df192866968d03569eb4da66b0fba8bfd6/dex-net/apps/generate-dataset-canny.py#L133

Then in the data loader, the grasp data is read and the second term is `level_score`, which should be the `miu` during generation. 
https://github.com/lianghongzhuo/PointNetGPD/blob/5d3453df192866968d03569eb4da66b0fba8bfd6/PointNetGPD/model/dataset.py#L799
Could you explain where did you change the it to `1/miu`?

lianghongzhuo, 2021-04-09:
Sorry I forget the detail of the implementation. In the implementation, the lower the value the better the quality.
see the code:
https://github.com/lianghongzhuo/PointNetGPD/blob/master/PointNetGPD/model/dataset.py#L537-L543
We did not use 1/miu in the labeling process.

Guptajakala, 2021-04-09:
I see, thanks for the clarification!

closed 2021-04-09 by Guptajakala