lianghongzhuo/PointNetGPD #40: What's the train data size on YCB if follow your pipeline?

Guptajakala, 2021-04-16:
Hi, I'm following your data generatoin pipeline on YCB objects in readme. Could you let me know when we run ``main_1v`` what's expected size of the ``PointGraspOneViewDataset`` dataset? 
https://github.com/lianghongzhuo/PointNetGPD/blob/5d3453df192866968d03569eb4da66b0fba8bfd6/PointNetGPD/main_1v.py#L117

Just want to make sure I'm not missing any step. It would be great for people to confirm this when following your work, if this can be written in readme.

lianghongzhuo, 2021-04-17:
Did you mean `grasp_points_num`? We tested several sizes. 500, 750, 1000... etc.

Guptajakala, 2021-04-17:
I mean the length of the ``PointGraspOneViewDataset``

For example, if I follow your train pipeline, what should be the output of this?
```
dataset = PointGraspOneViewDataset(...)
print(len(dataset))
```