lianghongzhuo/PointNetGPD #43: About your grasp dataset

pierhui, 2021-06-02:
Hi, Liang,
I have downloaded your 'PointNetGPD_grasps_dataset.zip',and I put it in the ~/code/PointNetGPD/PointNetGPD/data.Also I rename the PointNetGPD_grasps_dataset as ycb_grasp.But,when I python main_1v.py.I got the information as flowers.
[image]
Did I do it wrong?
By the way,I didn't generate my Own Grasp Dataset as written in the README.md.So I really want to know how I should use your generated dataset.

chenyunbo-cyb, 2022-01-24:
I have this same problem too!!!Help!!!

xiaofeiso, 2022-11-18:
> I have this same problem too, do you know how to solve it?

zundongke, 2023-03-13:
> > I have this same problem too, do you know how to solve it?

do you solve it?

zundongke, 2023-03-13:
> I have this same problem too!!!Help!!!

do you solve it ?

zundongke, 2023-03-13:
> Thank you very much!
do you solve it? i also meet the same problem.

xmkkkkkk, 2023-05-30:
> do you solve it ?

HI,do you solve it ?

lianghongzhuo, 2023-05-30:
I have updated the readme and the code. Please check, and if not work, please open a new issue.