lianghongzhuo/PointNetGPD #45: error:Install python pcl library python-pcl

quyouyuan, 2021-07-19:
Hello!liang! When I use your 'Python PCL' library to install, I report an error! Can you tell me how to solve it? The error is as follows:
setup.py: error: cannot find PCL, tried
    pkg-config pcl_common-1.9
    pkg-config pcl_common-1.8
    pkg-config pcl_common-1.7
    pkg-config pcl_common-1.6
    pkg-config pcl_common

aniketghodake10, 2022-02-09:
> Hello!liang! When I use your 'Python PCL' library to install, I report an error! Can you tell me how to solve it? The error is as follows: setup.py: error: cannot find PCL, tried pkg-config [...]

#47

lianghongzhuo, 2023-06-08:
I have switched the code to open3d

closed 2023-06-08 by lianghongzhuo