lianghongzhuo/PointNetGPD #47: python-pcl library installation error

aniketghodake10, 2021-12-14:
Ubuntu 20.04, conda environment python 3.7 

python setup.py build_ext -i

setup.py: error: cannot find PCL, tried
    pkg-config pcl_common-1.9
    pkg-config pcl_common-1.8
    pkg-config pcl_common-1.7
    pkg-config pcl_common-1.6
    pkg-config pcl_common

chenyunbo-cyb, 2022-01-22:
I have this problem too!!Help!!

aaronhd, 2022-02-08:
you need to install PCL lib firstly and then install python module.
`sudo apt-get install libpcl-dev -y` referring to https://github.com/lianghongzhuo/python-pcl

aaronhd, 2022-02-08:
> Ubuntu 20.04, conda environment python 3.7
> 
> python setup.py build_ext -i
> 
> setup.py: error: cannot find PCL, tried pkg-config pcl_common-1.9 pkg-config pcl_common-1.8 pkg-config pcl_common-1.7 pkg-config pcl_common-1.6 pkg-config pcl_common

you need to install PCL lib firstly and then install python module.
sudo apt-get install libpcl-dev -y referring to https://github.com/lianghongzhuo/python-pcl

aniketghodake10, 2022-02-08:
> you need to install PCL lib firstly and then install python module. sudo apt-get install libpcl-dev -y referring to https://github.com/lianghongzhuo/python-pcl

Ubuntu 20.04.....using python virtual env with python version 3.8.10.......
vtk 9.1 latest (installed via pip)......installed libpcl-dev also

still same error upon performing setup.py build

aniketghodake10, 2022-02-09:
SOLVED ubuntu 20.04

created conda environment with python=3.6 numpy=1.19.2 cython and activate the env.
sudo apt-get update
sudo apt-get install pcl-tools
sudo apt-get install libpcl-dev -y
conda config --add channels conda-forge
conda install -c sirokujira python-pcl
conda install -c jithinpr2 gtk3
conda install -y ipython

thnx aaronhd for the reply tho

closed 2022-02-09 by aniketghodake10

Xxianggg, 2022-04-10:
> SOLVED ubuntu 20.04
> 
> created conda environment with python=3.6 numpy=1.19.2 cython and activate the environment sudo apt-get update sudo apt-get install pcl-tools sudo apt-get install libpcl-dev -y conda config --add channels conda-forge conda install -c sirokujira python-pcl conda install -c jithinpr2 gtk3 conda install -y ipython
> 
> Then, go to your anaconda environment lib folder ( cd ~/anaconda3/envs/YOUR_ENV_NAME/lib) and execute following 4 commands: ln -s libboost_system.so.1.64.0 libboost_system.so.1.54.0 ln -s libboost_filesystem.so.1.64.0 libboost_filesystem.so.1.54.0 ln -s libboost_thread.so.1.64.0 libboost_thread.so.1.54.0 ln -s libboost_iostreams.so.1.64.0 libboost_iostreams.so.1.54.0
> 
> thanks @aaronhd for the reply tho

Hi, I've done the above steps, but there are still initial problems. I really don't know how to solve this.

JackeyLa5, 2022-04-16:
> 

Me too. Have you solved this problem?

lucasbrd, 2022-05-09:
Hey @Xxianggg @JackeyLa5 have you managed to find a solution ?
Thanks !

JackeyLa5, 2022-05-10:
> Hey @Xxianggg @JackeyLa5 have you managed to find a solution ? Thanks !

My problem is Ubuntu 20 04 through apt get install, pcl1.10 is installed by default , but the configuration file of python-pcl does not support a later version of PCL, so I uninstall pcl1.10, and install pcl1.9.1 through the source code, and then solved this problem.