lianghongzhuo/PointNetGPD #48: read_grasps_from_file.py mayavi qt4 error

aniketghodake10, 2022-02-11:
Ubuntu 20.04 python 3.6 conda environment

Same issue like this https://askpythonquestions.com/2022/02/06/python-loading-mayavi-mlab-throws-attribute-error-probably-from-qt-loading-mayavi-works-fine/

Any solution ??

penghongguang, 2022-03-20:
Have you solved it？I  meet the same problem.

penghongguang, 2022-03-20:
> Ubuntu 20.04 python 3.6 conda environment
> 
> Same issue like this https://askpythonquestions.com/2022/02/06/python-loading-mayavi-mlab-throws-attribute-error-probably-from-qt-loading-mayavi-works-fine/
> 
> Any solution ??

Have you solved it？I meet the same problem.

aniketghodake10, 2022-03-20:
> > Ubuntu 20.04 python 3.6 conda environment
> > Same issue like this https://askpythonquestions.com/2022/02/06/python-loading-mayavi-mlab-throws-attribute-error-probably-from-qt-loading-mayavi-works-fine/
> > Any solution ??
> 
> Have you solved it？I meet the same problem.

not yet

xiaofeiso, 2022-10-15:
you can see this:https://github.com/wkentaro/labelme/issues/842,but i have another problems.

lianghongzhuo, 2023-05-16:
if you use conda, you can try `conda install mayavi -c conda-forge`
my system is python3.10, and there is no problem

closed 2023-05-16 by lianghongzhuo