lianghongzhuo/PointNetGPD #51: AttributeError: 'tuple' object has no attribute 'center'

xiaofeiso, 2022-11-05:
when i run python generate-dataset-canny.py
I have this problem:
WARNING:root:Failed to import gqcnn! Grasp2D functions will not be available.
WARNING:root:Failed to import OpenRAVE
WARNING:dexnet.grasping.grasp_sampler:Failed to import rospy, you can't grasp now.
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
a worker of task  start
Log: do job 0
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Log: opened object 1 
INFO:dexnet.grasping.grasp_sampler:Num surface: 36266
Process Process-1:37:
Traceback (most recent call last):
  File "/home/xiaofeisong/anaconda3/envs/pointnetgpd/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/xiaofeisong/anaconda3/envs/pointnetgpd/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "generate-dataset-canny.py", line 111, in worker
    vis=False, random_approach_angle=True)
  File "/home/xiaofeisong/code/dex-net/src/dexnet/grasping/grasp_sampler.py", line 215, in generate_grasps
    dist = ParallelJawPtGrasp3D.distance(cur_grasp, grasp)
  File "/home/xiaofeisong/code/dex-net/src/dexnet/grasping/grasp.py", line 229, in distance
    center_dist = np.linalg.norm(g1.center - g2.center)
AttributeError: 'tuple' object has no attribute 'center'
Process Process-1:49:
Traceback (most recent call last):
  File "/home/xiaofeisong/anaconda3/envs/pointnetgpd/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/xiaofeisong/anaconda3/envs/pointnetgpd/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "generate-dataset-canny.py", line 111, in worker
    vis=False, random_approach_angle=True)
  File "/home/xiaofeisong/code/dex-net/src/dexnet/grasping/grasp_sampler.py", line 215, in generate_grasps
    dist = ParallelJawPtGrasp3D.distance(cur_grasp, grasp)
  File "/home/xiaofeisong/code/dex-net/src/dexnet/grasping/grasp.py", line 229, in distance
    center_dist = np.linalg.norm(g1.center - g2.center)
AttributeError: 'tuple' object has no attribute 'center'

xiaofeiso, 2022-11-05:
> when i run python generate-dataset-canny.py I have this problem: WARNING:root:Failed to import gqcnn! Grasp2D functions will not be available. WARNING:root:Failed to import OpenRAVE [...]

xiaofeiso, 2022-11-05:
I know what happen, it is my error, I used another dexnet.

closed 2022-11-05 by xiaofeiso

xmkkkkkk, 2023-05-25:
Hi，I encountered the same problem, how did you solve it？

lianghongzhuo, 2023-05-25:
did you use dexnet in this repo?

xmkkkkkk, 2023-05-26:
> 
Thank you for your prompt reply. This issue has been resolved,I used another dex-net.But when I run  python ycb_cloud_generate.py，I got new error. How can i solve it？
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/home/xmk/.conda/envs/py3/lib/python3.10/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/xmk/.conda/envs/py3/lib/python3.10/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
  File "/home/xmk/code/PointNetGPD/PointNetGPD/ycb_cloud_generate.py", line 318, in generate
    viewpoint_angle = path[2].split("_")[1].split(".")[0]  # Relative angle of the object w.r.t the camera
IndexError: list index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/xmk/code/PointNetGPD/PointNetGPD/ycb_cloud_generate.py", line 394, in <module>
    main()
  File "/home/xmk/code/PointNetGPD/PointNetGPD/ycb_cloud_generate.py", line 390, in main
    pool.map(generate, fl)
  File "/home/xmk/.conda/envs/py3/lib/python3.10/multiprocessing/pool.py", line 367, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
  File "/home/xmk/.conda/envs/py3/lib/python3.10/multiprocessing/pool.py", line 774, in get
    raise self._value
IndexError: list index out of range

xmkkkkkk, 2023-05-26:
My goodness, you updated this part of the code before, haha,