lianghongzhuo/PointNetGPD #54: 【train error】_verify_batch_size

xins981, 2022-12-17:
error：Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
Expected more than 1 value per channel when training, got input size torch.Size([1, 512])

this error is caused by the following sentence(pointnet.py line 35):
x = F.relu(self.bn4(self.fc1(x)))

@lianghongzhuo  hello, based my understanding, after 3 conv layer，pooling all point toward each feature channel. this seems reasonable. why throwing the error? How I should do to avoid this error?

nuo-code, 2023-02-20:
您好，我也遇到这个错误了，请问您解决了吗？能向您请教一下吗？谢谢

lianghongzhuo, 2023-06-09:
Thank you for pointing this out, I will add a note in the readme.

closed 2023-06-09 by lianghongzhuo

Legendaryaaa, 2023-06-09:
您发送的文件已收到~