lianghongzhuo/PointNetGPD #56: Read_file_sdf.py ,no result

jinxvyang, 2023-05-21:
Hello,Mr.liang,
when I generate nontextured.sdf file and nontextured.obj file using pcl-tools and SDFGen,
no error is reported when running  read_file_sdf.py  , but no results are output. 
I think there may be a problem with the version.
So shall you please provide the version of the dependencies in the requirements .txt ？

lianghongzhuo, 2023-05-21:
I guess your ycb data is not in the correct path, please download the newest code and run this file again.
If you see this error message: `[ERROR] There is no file in {file_dir}, Please check your data folder`, then you need to change your data path according to the readme.
If not please put your terminal output here.

jinxvyang, 2023-05-22:
Sorry,but there is no such error, it's just that there is no output in the terminal at that step.
I think it should be caused by some accumulated warnings during the previous setup compilation.
​So I think there may be problems with the version of the dependencies in the requirements.txt.

lianghongzhuo, 2023-05-22:
OK, you can use this as a reference, but no need to be strict on the version.
This is the version numbers from a newly installed conda environment.
```
# This file may be used to create an environment using:
# $ conda create --name <env> --file <this file>
# platform: linux-64
_libgcc_mutex=0.1=conda_forge
_openmp_mutex=4.5=2_gnu
addict=2.4.0=pypi_0
aiohttp=3.8.4=py310h1fa729e_0
aiosignal=1.3.1=pyhd8ed1ab_0
alsa-lib=1.2.8=h166bdaf_0
aom=3.5.0=h27087fc_0
apptools=5.1.0=pyh44b312d_0
asttokens=2.2.1=pyhd8ed1ab_0
async-timeout=4.0.2=pyhd8ed1ab_0
attr=2.5.1=h166bdaf_1
attrs=23.1.0=pyh71513ae_1
autolab-core=1.1.1=pypi_0
autolab-perception=1.0.0=pypi_0
backcall=0.2.0=pyh9f0ad1d_0
backports=1.0=pyhd8ed1ab_3
backports.functools_lru_cache=1.6.4=pyhd8ed1ab_0
bitarray=2.7.3=pypi_0
blinker=1.6.2=pypi_0
blosc=1.21.3=hafa529b_0
boost=1.82.0=py310h5a539fb_0
boost-cpp=1.82.0=he19a28c_1
brotli=1.0.9=h166bdaf_8
brotli-bin=1.0.9=h166bdaf_8
bzip2=1.0.8=h7f98852_4
c-ares=1.18.1=h7f98852_0
ca-certificates=2023.5.7=hbcca054_0
cairo=1.16.0=h35add3b_1015
catkin-pkg=0.5.2=pypi_0
certifi=2023.5.7=pyhd8ed1ab_0
chardet=5.1.0=pypi_0
charset-normalizer=2.1.1=pyhd8ed1ab_0
click=8.1.3=pypi_0
colorama=0.4.6=pypi_0
colorlog=6.7.0=pypi_0
comm=0.1.3=pypi_0
configargparse=1.5.3=pypi_0
configobj=5.0.8=pyhd8ed1ab_0
contourpy=1.0.7=py310hdf3cbec_0
curl=8.0.1=h588be90_0
cvxopt=1.3.1=pypi_0
cycler=0.11.0=pyhd8ed1ab_0
cython=0.29.34=pypi_0
dash=2.9.3=pypi_0
dash-core-components=2.0.0=pypi_0
dash-html-components=2.0.0=pypi_0
dash-table=5.0.0=pypi_0
dbus=1.13.6=h5008d03_3
debugpy=1.6.7=pypi_0
decorator=5.1.1=pyhd8ed1ab_0
dex-net=0.2.0=dev_0
dill=0.3.6=pypi_0
distro=1.8.0=pypi_0
dnspython=2.3.0=pypi_0
docutils=0.20=pypi_0
double-conversion=3.2.0=h27087fc_1
eigen=3.4.0=h4bd325d_0
envisage=6.0.1=pyhd8ed1ab_0
executing=1.2.0=pyhd8ed1ab_0
expat=2.5.0=hcb278e6_1
fastjsonschema=2.16.3=pypi_0
ffmpeg=5.1.2=gpl_h8dda1f0_106
ffmpeg-python=0.2.0=pypi_0
flann=1.9.1=h0224099_1012
[... 254 lines cut ...]
xkeyboard-config=2.38=h0b41bf4_0
xorg-fixesproto=5.0=h7f98852_1002
xorg-inputproto=2.3.2=h7f98852_1002
xorg-kbproto=1.0.7=h7f98852_1002
xorg-libice=1.0.10=h7f98852_0
xorg-libsm=1.2.3=hd9c2040_1000
xorg-libx11=1.8.4=h0b41bf4_0
xorg-libxau=1.0.9=h7f98852_0
xorg-libxdmcp=1.1.3=h7f98852_0
xorg-libxext=1.3.4=h0b41bf4_2
xorg-libxfixes=5.0.3=h7f98852_1004
xorg-libxi=1.7.10=h7f98852_0
xorg-libxrender=0.9.10=h7f98852_1003
xorg-libxt=1.2.1=h7f98852_2
xorg-renderproto=0.11.1=h7f98852_1002
xorg-xextproto=7.3.0=h0b41bf4_1003
xorg-xf86vidmodeproto=2.3.1=h7f98852_1002
xorg-xproto=7.0.31=h7f98852_1007
xxhash=3.2.0=pypi_0
xz=5.2.6=h166bdaf_0
yarl=1.9.1=py310h2372a71_0
zipp=3.15.0=pyhd8ed1ab_0
zlib=1.2.13=h166bdaf_4
zstd=1.5.2=h3eb15da_6
```

lianghongzhuo, 2023-06-08:
close due to no responds

closed 2023-06-08 by lianghongzhuo