lianghongzhuo/PointNetGPD #57: error：python main_1v.py --epoch 200 --mode train --batch-size 64

xmkkkkkk, 2023-05-31:
Hello Mr. Liang, I re-downloaded your PointNetGPD, but when I run 'python main_1v.py --epoch 200 --mode train --batch-size  64’, there is such an error.I don't launch a tensorboard for monitoring, I don't know if this is the reason. And I changed ‘--batch-size’ but still the same error. Here is my dataset path and the error. Thanks!
[image]
[image]

xmkkkkkk, 2023-05-31:
The python3.10 I used at first also had this error.Then I installed a lower version of pytorch with pyhton3.7 env and this error is also.

lianghongzhuo, 2023-06-08:
I am writing patches to fix this, the dataset path has changed. Please put your ycb object dataset (rgbd and google 512k) at `$PointNetGPD_FOLDER//data/ycb-tools/models/ycb`

lianghongzhuo, 2023-06-09:
this issuse is solved by https://github.com/lianghongzhuo/PointNetGPD/commit/3b53a495b8f730543f273504df4e5d27c3699572

closed 2023-06-09 by lianghongzhuo

xmkkkkkk, 2023-06-12:
Dear,Mr. Liang, Thank you for your answer, but now there is a new problem. I want to know at which step the file it said was generated. I really don’t have this file.
[image]

lianghongzhuo, 2023-06-12:
Hi, this is because of the missing model folder, see 336eb9e9cbb534dea36b3e92e0d1b7adef13d7f3 for the up to date code