lianghongzhuo/PointNetGPD #58: some questions about 'cam_pos' in kinect2grasp.py

xmkkkkkk, 2023-06-15:
Hello Mr. Liang, thank you for your PointNetGPD.I have some doubts on line 406 in kinect2grasp.py about 'cam_pos'.Do you mean I need to use tf to get the position between ‘/table_top’ and ‘/kinect2_ir_optical_frame’ or something else? But tf  doesn't seem to work properly in python3.  I'm sorry that my similar knowledge understanding is very shallow, if you can, could you please explain me in detail what do i need to change 'cam_pos' into and how to do it?Thanks again, looking forward to your reply

lianghongzhuo, 2023-06-15:
You can use ros-noetic, it support python3

xmkkkkkk, 2023-06-16:
Hi Mr. Liang,I catkin_make the source code of the tf package with python3, and import tf will not report the error. I have encountered a new problem. I only have the realsense camera, where do I change the points cloud topic of the kinect camera to realsense points cloud topic in kinect2grasp.py?

lianghongzhuo, 2023-06-17:
This is the technical question related to the basic use of ROS, I recommend you ask these questions at https://answers.ros.org/questions/ or search online.

closed 2023-06-17 by lianghongzhuo